georgewhewell/undervolt #1: Table with support testing

oscardssmith, 2017-12-30:
It would be good to have a table for what configurations this is known to work for.

georgewhewell, 2017-12-31:
Thanks, have added my config. If you can confirm working on your system please send PR with config :)

closed 2017-12-31 by georgewhewell

oscardssmith, 2017-12-31:
How can I confirm that it worked? It didn't produce any output suggesting an error, but I don't know how to see what the voltage is.

georgewhewell, 2017-12-31:
If it worked your temps/power usage should be a bit lower, and maybe your
system locks up if you apply too much undervolt! There is no 'read'
functionality yet- I will try to add this next.

AndreaOrru, 2018-04-16:
Hey @georgewhewell thanks for this! Tested on a `Lenovo Yoga 920 - i7-8550U` - it works :)

georgewhewell, 2018-04-17:
@AndreaOrru thanks- added!

commit 718216cb (2017-12-31): add hardware support table. resolves #1