georgewhewell/undervolt #12: Stopped working after kernel upgrade 4.15.0-22-generic Ubuntu 18.04 LTS

j0rd, 2018-05-28:
First of all, thank you for this undervolt script. I've been using it to lower the temps on my thinkpad x1 c6 laptop.

With that said, I upgraded my linux kernel last week and it appears to have broken the undervolt script unfortunately.

Kernel is from Ubuntu 18.04 .  4.15.0-22-generic #24-Ubuntu SMP Wed May 16 12:15:17 UTC 2018

Here's the error:

```
**root@x1c6#** ./undervolt.py -r
Traceback (most recent call last):
  File "./undervolt.py", line 243, in <module>
    main()
  File "./undervolt.py", line 238, in main
    voltage = read_offset(plane)
  File "./undervolt.py", line 159, in read_offset
    write_msr(value_to_write)
  File "./undervolt.py", line 38, in write_msr
    f = os.open(c, os.O_WRONLY)
OSError: [Errno 6] No such device or address: '/dev/cpu/15/msr'
```
Here's some text commands:

```
**root@x1c6#** ls -ltrh /dev/cpu/15/msr
crw------- 1 root root 202, 15 May 28 16:56 /dev/cpu/15/msr

**root@x1c6#** cat /dev/cpu/15/msr
cat: /dev/cpu/15/msr: No such device or address
```

georgewhewell, 2018-05-28:
Thanks for report. Seems strange. Can you post `ls -la /dev/cpu` please? Your CPU appears to have 8 threads, so I don't know why there are 16 /dev/cpu/ entries! What was the last kernel version that this worked on?

j0rd, 2018-05-29:
After a reboot, I'm able to get it working as expected. So this should be a none-issue and is most likely something wrong with something else.

As for your request, here's the output now, where undervolt is working correctly.

```
j0rd@x1c6:~/src/undervolt$ ls -la /dev/cpu
total 0
drwxr-xr-x 10 root root     220 May 28 23:59 .
drwxr-xr-x 19 root root    4260 May 28 23:57 ..
drwxr-xr-x  2 root root      60 May 28 23:59 0
drwxr-xr-x  2 root root      60 May 28 23:59 1
drwxr-xr-x  2 root root      60 May 28 23:59 2
drwxr-xr-x  2 root root      60 May 28 23:59 3
drwxr-xr-x  2 root root      60 May 28 23:59 4
drwxr-xr-x  2 root root      60 May 28 23:59 5
drwxr-xr-x  2 root root      60 May 28 23:59 6
drwxr-xr-x  2 root root      60 May 28 23:59 7
crw-------  1 root root 10, 184 May 28 23:57 microcode
```

georgewhewell, 2018-05-29:
Even stranger! I'll leave this open for a while in case anyone else reports this, otherwise looks unrelated.

closed 2018-06-08 by georgewhewell

freesamael, 2018-07-21:
I encountered the same issue today, that a 6C12T i7-8700 somehow shows 16 `/dev/cpu` entries. I simply changed the script to rely on `multiprocessing.cpu_count()` instead.

After a few reboots I noticed those "ghost" entries are gone. Probably it's MSR driver to blame?

georgewhewell, 2018-07-22:
Strange- I guess you are right, problem with driver. Can you make a PR with your change?

j0rd, 2018-11-05:
updated to latest and this resolves this issue.