georgewhewell/undervolt #14: Making change permanent?

dejIO, 2018-06-08:
What's the best way to make the change permanent? Using Arch Linux.

georgewhewell, 2018-06-10:
There is no way to make the change permanent in the sense of not having to run some utility- it will always be reset on reboot, suspend etc. There are some suggestions in #10 for re-applying changes automatically, or you can just run the command in a loop:

`watch -n 1 "undervolt --gpu -100"`

dejIO, 2018-06-11:
This is what I did. I think it's working.

```
/etc/systemd/system/undervolt.service

[Unit]
Description=undervolt

[Service]
Type=oneshot
RemainAfterExit=true
ExecStart=/usr/bin/sudo undervolt -v --core -150 --cache -150 --gpu -100
[Install]
WantedBy=multi-user.target
```

Xentradi, 2018-06-12:
Forgive me if this is the wrong way to go about it but couldn't it run with a cronjob in `crontab -e`?

`@reboot /path/to/undervolt.py -options`

If this is incorrect what would then be the most effective way?

georgewhewell, 2018-06-12:
@Xentradi This should work for applying undervolt after boot, but I think you will find after suspend/resume the settings need to be re-applied, and cron will not do this.

Xentradi, 2018-06-12:
@georgewhewell I noticed that after my comment. And have been working on a systemd based solution building upon what @dejIO used.

[https://gitlab.com/xentradi/undervoltWatcher](https://gitlab.com/xentradi/undervoltWatcher)
Probably going to move it to github if GitLab keeps giving me 502 errors.

Combines a systemctl service with a watch script.
Service ensures the watcher is always running even if killed unless the service is stopped.
Watcher checks (every 5 seconds) to see if the core offset values are 0.0 if they are it applies the offsets. 
Of course this only works if you're changing your core offset but it can be changes pretty easily.

Also thank you for providing undervolt. It's by far the simplest solution for under-volting!

georgewhewell, 2018-06-14:
Cool! Is there any reason you don't just apply the offset on every invocation? 'Reading' still involves writing to the register and reading back result, so the only difference is whether the 'read' bit is set.

Xentradi, 2018-06-15:
Mostly for convenience, The values I have set are what I know are stable right now. I like to try different offsets to see what sort of results I can get. If it was setting at every invocation my tests would last all of 5 seconds; this way it will go till either I change it or the system sleeps/reboots.

Oliph, 2018-06-27:
To my system service file I added `Timer` because I set up wrong value once and I had kernel panic at the boot. Here I am sure I can fix it after the boot. 

Also, quick question. Why do you need an undervolt-watcher? Does the values get reset randomly?

```
[Unit]
Description=Undervolting the CPU and the GPU
After=systemd-user-sessions.service

[Timer]
OnBootSec=2min
[Service]
ExecStart=/usr/bin/undervolt --core -105 --cache -105 --gpu -80

[Install]
WantedBy=multi-user.target
```

For compatibility purpose, it is for the Lenovo X1Y3 and the i7-8550

Xentradi, 2018-06-27:
@Oliph your offsets will reset on boot, and on wake from sleep/hibernation. I use a laptop so my offsets were getting reset when ever the lid was closed. The watcher corrects this while allowing me to still tweak the offsets when I want and I can reset them to something I know works as simply as closing the lid. I use that mostly because my schedule is unpredictable so I'm not always able to sit and test

commit 942ce311 (2018-07-24): add instructions for creating systemd units. fixes #10, fixes #14
closed 2018-07-24 by georgewhewell