georgewhewell/undervolt #20: [Feature request] Different temperature target throttling when running on battery

mensfeld, 2018-07-31:
Would it be doable to provide an option to change the temperature target throttling level when running on battery? When on battery I would prefer to lower it to around 75 in order to extend time over performance.

georgewhewell, 2018-07-31:
Good idea- should be possible. Maybe one problem is that I've seen my machine crash when setting the temp limit to something lower than the current temperature. Consider the scenario where your machine is on AC running something intensive, core temp is 95C. Then you unplug AC- your cpu is 20C over critical limit, I think this causes a reset. Maybe check if you can replicate this?

mensfeld, 2018-07-31:
@georgewhewell thank you for getting back to me. In a case like that, I would expect the system to gradually go down within the limit. So for your case it would be:

- plugged - 93 / limit 95
- plugged - 95 / limit 95
- unplugged - 95 / limit 95
- unplugged - 93 / limit 93
- unlugged - 88 / limit 88
...
- unlugged 75 / limit 75

I will give it a shot later during the day.

mensfeld, 2018-07-31:
@georgewhewell thanks for a reset ;) I guess the only idea I would have is to have something like stated above. If there is a limit over which it restarts (let's say the base +5) it should be enough as it would be throttled constantly lowering the temp.

georgewhewell, 2018-07-31:
good to know it's not just me :)

After experimenting a bit I'm not sure now this is to do with temperature limits- I've been running the following simultaneously with mprime torture test and i7z (to observe temperatures):

`while true; do sudo undervolt -r --temp 60 && sleep 5 && sudo undervolt -r --temp 95 && sleep 5; done`

I get a reset pretty quickly with my normal undervolt settings applied, but I haven't observed a reset with stock voltage settings. Perhaps this is just very stressful for CPU, and requires dialing back the undervolt settings a bit?

mensfeld, 2018-07-31:
I've been playing with it with following settings:

```
undervolt --gpu -75 --core -100 --cache -100 --uncore -100 --analogio -100
```

With `120` for core and cache my system becomes unstable. What you're saying would mean, that it would be good to have different undervolt settings not only for the throttling but also for the rest of the values. That way it could be prevented.

georgewhewell, 2018-07-31:
Perhaps that could be a solution, but it makes things more complicated. I think probably the reset indicates that the undervolt is just too aggressive, and it's not truly 'stable' at those offsets- I suspect reset could be triggered in ways other than rapid temperature change such as full GPU utilization. Maybe we need a 'torture test' script to find lowest safe voltages, then add 10mV for safety.

Anyway, I added --temp-ac and --temp-bat in this branch: https://github.com/georgewhewell/undervolt/commits/feat/selective-temp-on-ac-status

Note that it's stateless, so you need to run it periodically to pick up ac status change, eg using systemd timer in README

mensfeld, 2018-07-31:
Yeah anyhow you've added a systemd integration si it should't be a problem :) Great job with this one really!

closed 2018-07-31 by georgewhewell

mensfeld, 2018-07-31:
Great job! Thank you!

commit 13739b78 (2018-07-31): allow setting different temperature targets on AC/battery

fixes #20