georgewhewell/undervolt #21: noob needs help

xxvinxx, 2018-07-31:
Hi, I did the installation but when I try running it said
"undervolt: command not found"
did I do something wrong?
it did said:
"Collecting undervolt
Installing collected packages: undervolt
Successfully installed undervolt-0.2.7"

georgewhewell, 2018-07-31:
Strange- if you run `pip install undervolt` again, does it say where it installed to? Perhaps try `~/.local/bin/undervolt`

xxvinxx, 2018-08-01:
it is works, thank you

xxvinxx, 2018-08-16:
hi,
I am trying it on another device this time.
it returns this:
`modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "/home/vin/.local/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/home/vin/.local/lib/python2.7/site-packages/undervolt.py", line 229, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1`

Hope you can help,
Thank you,
Vin

georgewhewell, 2018-08-16:
Did you try with `sudo undervolt`? maybe check running `sudo modprobe msr` before

xxvinxx, 2018-08-16:
`sudo undervolt` does not work.

`sudo modprobe msr` before `~/.local/bin/undervolt` give this error:
`OSError: [Errno 13] Permission denied: '/dev/cpu/0/msr'`
which is better than last time but still not working.
I don't really know how to grant that permission.

closed 2018-08-16 by xxvinxx

v-marinkov, 2018-10-24:
I have a similar problem. If I run `undervolt` without sudo, it only gives errors. If I run it with sudo, I get a temperature reading but also the same errors:
`viktor@pop-os:~$ sudo /home/viktor/apps/anaconda3/bin/undervolt -r`
`temperature target: -3 (97C)`
`Traceback (most recent call last):`
`  File "/home/viktor/apps/anaconda3/bin/undervolt", line 11, in <module>`
`    sys.exit(main())`
`  File "/home/viktor/apps/anaconda3/lib/python3.7/site-packages/undervolt.py", line 275, in main`
`    voltage = read_offset(plane)`
`  File "/home/viktor/apps/anaconda3/lib/python3.7/site-packages/undervolt.py", line 170, in read_offset`
`    write_msr(value_to_write)`
`  File "/home/viktor/apps/anaconda3/lib/python3.7/site-packages/undervolt.py", line 45, in write_msr`
`    os.write(f, pack('Q', val))`
`OSError: [Errno 5] Input/output error`