georgewhewell/undervolt #22: Inpuut/Output error on --read

StanislavNikolov, 2018-08-08:
```
stjo@swanky undervolt $ ./undervolt.py -r     
Traceback (most recent call last):
  File "./undervolt.py", line 285, in <module>
    main()
  File "./undervolt.py", line 276, in main
    tjunc=read_temperature(),
  File "./undervolt.py", line 157, in read_temperature
    return read_msr(0x1a2) >> 24
  File "./undervolt.py", line 54, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'
stjo@swanky undervolt $ sudo ./undervolt.py -r
temperature target: -0 (100C)
Traceback (most recent call last):
  File "./undervolt.py", line 285, in <module>
    main()
  File "./undervolt.py", line 280, in main
    voltage = read_offset(plane)
  File "./undervolt.py", line 170, in read_offset
    write_msr(value_to_write)
  File "./undervolt.py", line 45, in write_msr
    os.write(f, pack('Q', val))
OSError: [Errno 5] Input/output error
```
Here some (hopefully) useful info:
uname
```
stjo@swanky undervolt $ uname -a
Linux swanky 4.8.17-galliumos #1 SMP PREEMPT galliumos4 Thu Feb 23 02:27:28 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

/proc/cpuinfo
```
stjo@swanky undervolt $ cat /proc/cpuinfo 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 55
model name	: Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz
stepping	: 8
microcode	: 0x816
cpu MHz		: 499.000
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 4331.60
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 55
model name	: Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz
stepping	: 8
microcode	: 0x816
cpu MHz		: 833.000
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 4342.68
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```
I'm using a toshiba chromebook 2 (2014 edition, codename swanky), model CB35-B{4 numbers that I don't think anyone cares about}.

StanislavNikolov, 2018-08-08:
Output after adding `print(val, pack('Q', val))` above line 45:
```
stjo@swanky undervolt $ sudo ./undervolt.py -r
temperature target: -0 (100C)
9223374304597508096 b'\x00\x00\x00\x00\x10\x02\x00\x80'
Traceback (most recent call last):
  File "./undervolt.py", line 286, in <module>
    main()
  File "./undervolt.py", line 281, in main
    voltage = read_offset(plane)
  File "./undervolt.py", line 171, in read_offset
    write_msr(value_to_write)
  File "./undervolt.py", line 46, in write_msr
    os.write(f, pack('Q', val))
OSError: [Errno 5] Input/output error
```

georgewhewell, 2018-08-08:
See https://github.com/georgewhewell/undervolt#troubleshooting

Have you checked secure boot? I guess we can probably detect this and tell the user

v-marinkov, 2018-10-24:
I am facing the same issue on Thinkpad T430 with i7-3610qm. Secure boot and the security chip are turned off in the BIOS. is there any workaround?

georgewhewell, 2018-10-24:
I only just saw @StanislavNikolov is using Atom CPU- I think these are pretty different from Core series, probably don't have the same FIVR. I will update compatibility table to show these CPUs are not supported.

@v-marinkov Hm, it looks like your config should be working. Can you post distro and kernel version? I have a T520 I can test at the weekend. Does `dmesg | grep -i lockdown` show anything?

v-marinkov, 2018-10-24:
https://github.com/mihic/linux-intel-undervolt says only 
> Intel CPUs with an integrated voltage controller (FIVR)
are supported and FIVR was introduced in Haswell cpus (4th gen). Could this be the reason we get these errors on our Bay Trail and Ivy Bridge?

v-marinkov, 2018-10-24:
`dmesg | grep -i lockdown` shows nothing regardless of sudo.
Kernel is 4.15.0-38-generic
Pop!_OS 18.04 LTS

georgewhewell, 2018-10-24:
Ah, I think you are right- the earliest CPU this is confirmed to work on is i5-4xxx which is indeed Haswell. Thanks for report- you can make PR to add i7-3610 as unsupported if you want

closed 2018-10-29 by georgewhewell

NischayaSharma, 2020-03-18:
@georgewhewell 
I am facing literally the same issue, I have a Lenevo thinkpad X250, running intel i5-5300u.
on doing `dmesg | grep -i lockdown` i get the following result.
`dmesg | grep -i lockdown                                                                                                                                                                         
[    0.000000] Kernel is locked down from EFI secure boot; see man kernel_lockdown.7
[    1.268717] Lockdown: Hibernation is restricted; see man kernel_lockdown.7
[    4.391632] Lockdown: /dev/mem,kmem,port is restricted; see man kernel_lockdown.7
[   18.609578] Lockdown: ioperm is restricted; see man kernel_lockdown.7
[ 6799.454517] Lockdown: Direct MSR access is restricted; see man kernel_lockdown.7`

commit 340bd9a3 (2018-10-29): add unsupported CPU to hw table. closes #22