georgewhewell/undervolt #24: Adding TDP/Power limit

Oliph, 2018-08-14:
Hi, 
Are you planning to add option to play with the TDP? 

This wrapper is doing it: https://github.com/kitsunyan/intel-undervolt

georgewhewell, 2018-08-15:
I've thought a bit about this- mostly can't think of a nice way to expose options for user. I think it would be nice to allow different settings on battery and AC (like temp limit), so we 8 values in total to set. Maybe if we allow two values per option we can have --tdp-short-bat "15 30" --tdp-long-ac "20 40" etc. Any better suggestion? PR welcome if you want to attempt this yourself :)

Oliph, 2018-08-15:
For the option (and could be applied for the temperature too) I have a suggestion, while not solving all issues, remove at least one option. 

Rather than having one option for all, one option for AC and one option for battery why not just applying the temperature for ac and bat automatically. If the user also use a bat, the value is overridden when it is on battery. Then for instance for temperature.

Case 1: Uses of  -t  only
* The temperature is applied for both AC and BAT (like now)

Case 2: Use of -t and -t-bat
* the temperature are different for -t and -t-bat and use the values inserted. The -t is only used for AC

Case 3: Use of -t-bat only
* the temperature remains unchanged for the AC (-15) and change for battery only.

For PR, I may take a look but I am far competent for that.

georgewhewell, 2018-08-16:
@Oliph That's a good suggestion, thanks! I think I will have some time this weekend to look at this

georgewhewell, 2020-06-08:
Closed via #93

closed 2020-06-08 by georgewhewell