georgewhewell/undervolt #25: XPS 9570 i9-8950HK - Weird CPU Temp Target

berglh, 2018-08-15:
Seemed to report correct underclock voltages, just a werid temperature target:
```
temperature target: -131 (-31C)
core: -99.61 mV
cache: -99.61 mV
analogio: -99.61 mV
uncore: -99.61 mV
gpu: -75.2 mV
```

sensors shows this:
```
Package id 0:  +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
```

ties, 2018-08-15:
Same applies for HP Zbook Studio G5 with i7-8750H. Based on the readings I can confirm that the tool is working on this platform. However, I have not checked it's influence on power usage.

Platform:
```
$ sudo dmidecode | grep "Product Name" | head -n 1
	Product Name: HP ZBook Studio G5
$ cat /proc/cpuinfo| grep 8750 | head -n 1        
model name	: Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
```
Readings:
```
$ sudo `which python` undervolt.py -r
temperature target: -130 (-30C)
core: -99.61 mV
gpu: -75.2 mV
cache: -99.61 mV
uncore: -99.61 mV
analogio: -99.61 mV
```

With sensors reading correct range:
```
$ sensors | head -n 9                     
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
```

georgewhewell, 2018-08-15:
Can you post the output when `--verbose` flag set please?

berglh, 2018-08-15:
@georgewhewell As requested:

```
INFO:root:Read 0x83640000 from /dev/cpu/0/msr
INFO:root:Read 0x83640000 from /dev/cpu/0/msr
temperature target: -131 (-31C)
INFO:root:Writing 0x8000001000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/8/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/9/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/10/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/11/msr
INFO:root:Read 0xf3400000 from /dev/cpu/0/msr
core: -99.61 mV
INFO:root:Writing 0x8000021000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/8/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/9/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/10/msr
INFO:root:Writing 0x8000021000000000L to /dev/cpu/11/msr
INFO:root:Read 0xf3400000 from /dev/cpu/0/msr
cache: -99.61 mV
INFO:root:Writing 0x8000041000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/8/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/9/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/10/msr
INFO:root:Writing 0x8000041000000000L to /dev/cpu/11/msr
INFO:root:Read 0xf3400000 from /dev/cpu/0/msr
analogio: -99.61 mV
INFO:root:Writing 0x8000031000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/8/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/9/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/10/msr
INFO:root:Writing 0x8000031000000000L to /dev/cpu/11/msr
INFO:root:Read 0xf3400000 from /dev/cpu/0/msr
uncore: -99.61 mV
INFO:root:Writing 0x8000011000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/8/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/9/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/10/msr
INFO:root:Writing 0x8000011000000000L to /dev/cpu/11/msr
INFO:root:Read 0xf6600000 from /dev/cpu/0/msr
gpu: -75.2 mV
```

georgewhewell, 2018-08-15:
@berglh Thanks- seems like your platform is setting some high bits in the temp value which are not set on others, although we should be ignoring those anyway. Can you try branch `bug/fix-temp-parsing`?

ties, 2018-08-15:
Very similar output. Difference (in sorted output) with my i7-8750H is in temperature target and the accompanying read:
```
...
INFO:root:Read 0x83640000 from /dev/cpu/0/msr		      |	INFO:root:Read 0x82640000 from /dev/cpu/0/msr
INFO:root:Read 0x83640000 from /dev/cpu/0/msr		      |	INFO:root:Read 0x82640000 from /dev/cpu/0/msr
...
temperature target: -131 (-31C)				      |	temperature target: -130 (-30C)
...
```

georgewhewell, 2018-08-15:
@ties nice- the second output looks correct.

commit 4ed8830e (2018-08-15): discard high bits from temp reading. fixes #25
closed 2018-08-15 by georgewhewell