georgewhewell/undervolt #35: Remove cache option

oscardssmith, 2018-09-03:
since core and cache have to be set the same, why not just use core to set cache also?

georgewhewell, 2018-09-04:
Hm- not sure about this. AFAIK throttlestop and intel XTU have distinct settings for these (not sure though- never used either)- perhaps there is some CPU that exists (or will exist in the future) without this core == cache restriction.. I think we should retain the ability to read/write these separately. As a compromise, how about we automatically set cache to the same as cpu if it is not specified, instead of printing warning?

georgewhewell, 2018-09-04:
@mr700 Haha- well noticed! this is my mistake- i guess i was playing with voltages before capturing the -r output. My CPU does not return values like this normally :)

closed 2019-12-15 by georgewhewell