georgewhewell/undervolt #39: 9570 temperature target constant

eadwu, 2018-09-16:
Run `sudo undervolt -t 95` (or any number) and expect `sudo undervolt -r` to output correct temperature target.

`sudo undervolt -r`
```
temperature target: -3 (97C)
core: -99.61 mV
gpu: -75.2 mV
cache: -99.61 mV
uncore: -99.61 mV
analogio: -99.61 mV
```

`sudo undervolt -v -t 95`
```
INFO:root:Writing 0x5000000 to /dev/cpu/0/msr
INFO:root:Writing 0x5000000 to /dev/cpu/1/msr
INFO:root:Writing 0x5000000 to /dev/cpu/2/msr
INFO:root:Writing 0x5000000 to /dev/cpu/3/msr
INFO:root:Writing 0x5000000 to /dev/cpu/4/msr
INFO:root:Writing 0x5000000 to /dev/cpu/5/msr
INFO:root:Writing 0x5000000 to /dev/cpu/6/msr
INFO:root:Writing 0x5000000 to /dev/cpu/7/msr
INFO:root:Writing 0x5000000 to /dev/cpu/8/msr
INFO:root:Writing 0x5000000 to /dev/cpu/9/msr
INFO:root:Writing 0x5000000 to /dev/cpu/10/msr
INFO:root:Writing 0x5000000 to /dev/cpu/11/msr
```

Also tried disabling service and running undervolt alone, `sudo undervolt --core -100 --cache -100 --gpu -75 --uncore -100 --analogio -100 -t 95`.

georgewhewell, 2018-09-17:
Hm, that's strange. I noticed the same behavior on my desktop PC (and same fixed 97C limit). I wonder if it's specific to the platform/motherboard or the CPU itself. There are some other registers which say whether various functionality like changing voltage, power limit etc are allowed, perhaps there is one saying whether changing temperature limit is locked too.

eadwu, 2018-09-18:
#25 was a problem before this that I also experienced though it has now been resolved. Since they don't seem to have problems after that, I would assume that the temperature target worked for them. Though that shouldn't be the case since I'm also using an XPS 9570 with an i7-8750H.

georgewhewell, 2018-09-20:
Hm, extra strange. Perhaps there could be some setting in your BIOS that affect this?

eadwu, 2018-09-29:
Sorry for the late response, just rechecked the BIOS settings and I don't see any that seem to affect it.

eadwu, 2018-10-08:
Seems to work, might've been something in the BIOS since the only major things I've done was the Windows update and the BIOS update.

closed 2018-10-08 by eadwu