georgewhewell/undervolt #43: Cannot undervolt iGPU of i7-8705G

Element-126, 2018-10-16:
The CPU part of the i7-8705G can be successfully undervolted using `undervolt`, however the `--gpu` option does not seem to have any effect (although `undervolt --read` reports a non-zero offset).

Steps to reproduce:
---
1.  Undervolt the GPU with a sufficiently large (negative) offset:
    ```
    $ sudo undervolt --core -0 --cache -0 --uncore -0 --analogio -0 --gpu -500 --temp
    $ sudo undervolt --read
    ```
    Output:
    ```
    temperature target: -0 (100C)
    core: 0.0 mV
    cache: 0.0 mV
    analogio: 0.0 mV
    uncore: 0.0 mV
    gpu: -500.0 mV
    ```
2. Stress the GPU (e.g. run Unigine Superposition)

Expected behaviour:
---
* The GPU will likely crash immediately at step 1, and in any case the stress test should crash it.

Observed behaviour:
---
* Nothing happens. The stress test completes successfully. Temperature is not affected so the iGPU is probably drawing as much power as before.

System info:
---
* CPU: Intel® Core™ i7-8705G CPU @ 3.10GHz × 8
* iGPU: Intel® HD Graphics 630 (Kaby Lake GT2)
* Laptop: Dell XPS 9575
* Distribution: Fedora 28 (64 bit)
* Secure boot: off
* Kernel: 4.18.13-200.fc28.x86_64

georgewhewell, 2018-10-17:
Hm.. did you try some voltages in between 0 and -500?

Element-126, 2018-10-17:
Yes, I tried with a sequence of voltages from 0 to -200 mV, using -10 mV increments, and then a few more voltages below that. Note that at some point it seems to underflow (setting -1000 mV reports **+1000** mV when using `--read`, which could potentially be dangerous if this setting was actually having any effect)

gabrielspadon, 2019-03-14:
Hi there! 

I'm having the same problem on i7-8750H and it is quite awkward. However, I suspect that there might be some kind of fail-safe on the processor that prevents me from changing the GPU voltage — not only the GPU but also uncore and anologio. I decided to focus on core/cache (they are the same in this case) and tweak the threshold of my fans. Doing so, I was able to get lower temperatures, enough to avoid throttling on non-extreme computing task.

See the links above:

- [CPU_frequency_scaling](https://wiki.archlinux.org/index.php/CPU_frequency_scaling)
- [fan_speed_control](https://wiki.archlinux.org/index.php/fan_speed_control)