georgewhewell/undervolt #44: Unable to set it to start on boot

reikdas, 2018-11-14:
Undervolt is installed for root. Would that cause an issue? (I had to install it as root, otherwise it was not getting the appropriate permissions)

```
root@pratyush-GS65-Stealth-Thin-8RF:~# which undervolt
/usr/local/bin/undervolt
```
Here is my undervolt.service
```
pratyush@pratyush-GS65-Stealth-Thin-8RF:~$ cat /etc/systemd/system/undervolt.service
[Unit]
Description=undervolt

[Service]
Type=oneshot
# If you want to run from source:
ExecStart=/usr/bin/local/undervolt -v --core -180 --cache -180 --gpu -70
```

Error - 
```
root@pratyush-GS65-Stealth-Thin-8RF:~# systemctl start undervolt
Job for undervolt.service failed because the control process exited with error code.
See "systemctl status undervolt.service" and "journalctl -xe" for details.
```

georgewhewell, 2018-11-14:
Hm- what does it say in the logs? `sudo journalctl -xeu undervolt.service`

reikdas, 2018-11-15:
Seems like it was a false negative. Undervolt is working properly on boot.

closed 2018-11-15 by reikdas