georgewhewell/undervolt #45: Why run the timer every 30 seconds?

Utini2000, 2018-12-11:
I really like your script and how simple and effective it is.
How ever, running it every 30 seconds makes me wonder.

Wouldn't it be enough to just run it after booting and after comming back from suspend? 
Why would you run the script constantly every 30 seconds?

georgewhewell, 2018-12-12:
Hi, I think this was added because there are other events that trigger the system firmware to reset the voltages such as dock/undock (#10) which are not supported as triggers by systemd. This seemed simplest way to solve it

Utini2000, 2018-12-12:
Hey @georgewhewell ,
I understand that and already expected this to be the reason. 
But I find spamming the system with executing a script every 30 seconds a bit too much?
Isn't there some smarter solution which uses less resources? Maybe event monitoring?

toverux, 2019-01-03:
@JHrenyk would you like to share it? :)

JHrenyk, 2019-01-03:
I'd love to. I should be able to share it tonight. What's the best way to post it?

toverux, 2019-01-06:
Thanks a lot! Will try that!

> What's the best way to post it?

You could have just pasted the files here if you don't want to keep a repo for that :)

closed 2020-06-08 by georgewhewell