georgewhewell/undervolt #46: Error: could not insert 'msr'

Ostkreutz, 2019-01-06:
So I tried to use this in order to undervolt my i7-8750H as it puts out a little too much heat for my liking. However, when i type: 

`undervolt -r`

I get the following error:

`modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "/home/stefan/anaconda3/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/home/stefan/anaconda3/lib/python3.7/site-packages/undervolt.py", line 231, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/home/stefan/anaconda3/lib/python3.7/subprocess.py", line 341, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1.
`

I installed msr-tools and when i type `sudo undervolt -r` it tells me `sudo: undervolt: Command not found`. Does anybody have any idea on how to fix this? 
Any help is greatly appreciated.

fmellomascarenhas, 2019-01-11:
I used synaptic to install msr-tools

I also did: sudo modprobe msr

After that, the message changed to:
Traceback (most recent call last):
  File "/home/felipemello/anaconda3/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/home/felipemello/anaconda3/lib/python3.7/site-packages/undervolt.py", line 271, in main
    tjunc=read_temperature(),
  File "/home/felipemello/anaconda3/lib/python3.7/site-packages/undervolt.py", line 157, in read_temperature
    return (read_msr(0x1a2) & (127 << 24)) >> 24
  File "/home/felipemello/anaconda3/lib/python3.7/site-packages/undervolt.py", line 54, in read_msr
    f = os.open(n, os.O_RDONLY)
**PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'**

I believe it is because of the Secure Boot. I am thinking about disabling it, but I am not sure if it is unsafe

Hedronmx, 2019-01-17:
I already disabled secure boot and the same error occurs  
`PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'`

Alecppt, 2019-01-17:
one alternative is to use [intel-undervolt ](https://github.com/kitsunyan/intel-undervolt). I had the same error as well and disabling secure boot could not solve the problem. However, intel-undervolt works without msr permission error

georgewhewell, 2019-01-18:
try `sudo $(which undervolt) -r`

you don't need `msr-tools` or `modprobe msr` (script will do that for you)

pahennessey, 2019-01-31:
I'm trying to use this workaround  with the walkthrough to create a startup service, but I'm getting the following error

` ✘  ~  systemctl status undervolt.service
Warning: The unit file, source configuration file or drop-ins of undervolt.service changed on disk. Run 'systemctl daemo
● undervolt.service - undervolt
   Loaded: loaded (/etc/systemd/system/undervolt.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-01-30 19:07:37 CST; 4s ago
  Process: 5390 ExecStart=/usr/bin/sudo $(which undervolt) -v --core -150 --cache -150 --gpu -50 (code=exited, status=1/
 Main PID: 5390 (code=exited, status=1/FAILURE)

Jan 30 19:07:37 pop-os systemd[1]: Starting undervolt...
Jan 30 19:07:37 pop-os sudo[5390]: sudo: undervolt): command not found
Jan 30 19:07:37 pop-os systemd[1]: undervolt.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 19:07:37 pop-os systemd[1]: undervolt.service: Failed with result 'exit-code'.
Jan 30 19:07:37 pop-os systemd[1]: Failed to start undervolt.
lines 1-12/12 (END)
`

Any suggestions?

georgewhewell, 2019-01-31:
@pahennessey find the path to `undervolt.py` with `which undervolt` and put that path in the service file, don't use `which` in the service file.

patrick-toptal, 2019-02-01:
Ah, of course! The obvious answer! Thanks, @georgewhewell

pahennessey, 2019-02-01:
Of course, @georgewhewell , the obvious answer! Thanks!

sjpatel21, 2019-02-16:
I also got this error and I fixed by running it with sudo:
`sudo undervolt --read`

Coffeecoder, 2019-04-30:
Had the same error message as @Ostkreutz . 
`sudo undervolt --read` gave me the following error:
`sudo: undervolt: command not found`
moving the executable from `home/myname/.local/bin` to `/bin` solved the problem. Now I can run undervolt with sudo permissions. I still get the permission error when I run it without root permissions.

NicolasNewman, 2019-07-14:
@Coffeecoder Alternatively, you could install it via `sudo pip install undervolt`

gdav, 2019-07-25:
I had the same issue.
After disabling secure boot undervolt was working as expected again.
XPS 7590 with i7-9750H (please add to list of working devices ;) ).

But this is already noted in the README file.