georgewhewell/undervolt #49: undervolt --read command not found

philiporlando, 2019-02-08:
I'm getting stuck on something easy, but hopefully answering this question will help the other noobs out there!

I've cloned the git repo, but I don't have access to the `undervolt` command in the way that the documentation suggests. When I run `undervolt --read` I get the following error: 

`undervolt: command not found`.

I've also tried navigating to the `~/undervolt` directory and tried `python undervolt.py --read` which produces a different error:

```
modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "undervolt.py", line 283, in <module>
    main()
  File "undervolt.py", line 234, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1

```

Any insight is appreciated!

philiporlando, 2019-02-08:
Alright I figured out what I was doing wrong. So I was mistakenly using the `python2` version of `pip` during the `pip install undervolt`!  After `pip uninstall undervolt` and `pip3 install undervolt` I was able to access the `undervolt` module directly from the terminal. Afterwards, I was still faced with a modprobe error, but I realized that this has to do with permissions and is not a bug. Entering `sudo undervolt --read` did the trick!

closed 2019-02-08 by philiporlando

NickAlvesX, 2019-03-11:
I'm getting the following error:

error: invalid command 'bdist_wheel' 
  ----------------------------------------
  Failed building wheel for undervolt