georgewhewell/undervolt #51: Can't get it to start on boot

KaramAbuaisha, 2019-02-21:
Feb 21 01:11:29 idorobots sudo[15403]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/python3 /home/karam/.local/bin/undervolt -v --core -101 --cache -101 --gpu -75
Feb 21 01:11:29 idorobots sudo[15403]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 21 01:11:29 idorobots startupuv[15402]: Traceback (most recent call last):
Feb 21 01:11:29 idorobots startupuv[15402]:   File "/home/karam/.local/bin/undervolt", line 7, in <module>
Feb 21 01:11:29 idorobots startupuv[15402]:     from undervolt import main
Feb 21 01:11:29 idorobots startupuv[15402]: ImportError: No module named 'undervolt'
Feb 21 01:11:29 idorobots sudo[15403]: pam_unix(sudo:session): session closed for user root

KaramAbuaisha, 2019-02-21:
Issue fixed by pip installing undervolt as root

closed 2019-02-21 by KaramAbuaisha

kylebakerio, 2021-02-21:
for googlers: https://github.com/georgewhewell/undervolt/issues/63