georgewhewell/undervolt #52: Getting undervolt to work on Qubes?

dexinthecity, 2019-02-22:
Has anyone been successful in getting undervolt to work in Qubes OS? Trying to get my X1C6 undervolt to fix some heating issues.

georgewhewell, 2019-02-25:
Have not tried but I think it should work if you run it in dom0. Would be very surprised if you can access these MSR from within VM though

dexinthecity, 2019-02-26:
So it seems that since dom0 does not have network access I would need to compile it myself inside dom0...the problem is I don’t know what the heck I’m doing lol. Would you be able to help me compile this inside dom0?

georgewhewell, 2019-02-26:
You don't need to compile anything- you can just download `undervolt.py` and run it (provided you have python available). I'm not very familiar with qubes so can't offer any help!

closed 2019-12-15 by georgewhewell