georgewhewell/undervolt #58: LENOVO ThinkPad X1 Extreme 20MFCTO1WW i7-8750H CPU

retromuz, 2019-04-01:
I can confirm undervolt works awesome on above configuration. I could use below and get more than 10 hours (sometimes 12).

$ sudo undervolt --read
temperature target: -37 (63C)
core: -155.27 mV
cache: -155.27 mV
analogio: 0.0 mV
uncore: -155.27 mV
gpu: -155.27 mV

retromuz, 2019-04-01:
On high performance using stress test, it maxes out at 3.1GHz using 63 celcius temperature target even without turning the fans on. When it warms up it settles down at ~2GHz at 25 celcius ambient temperature. This is awesome. I have a fast machine which is completely silent.

wokawoka, 2019-04-02:
Impressive. 
Since there are many combinations of 20MFCTO1WW, does your machine drive a FHD (1920x1080) or a 4K UHD HDR (3840x2160) screen?

retromuz, 2019-04-02:
Mine is FHD, which I find completely satisfying experience compared to 2011 MBP I replaced with this. Keep in mind this weights slightly less as well.

BTW, I used **stress** to stress test the machine.
I use Pop!OS by system76 daily, which is damn good and supports dual graphics without a hitch! I highly recommend you guys give it a shot. It is so lean and clean.

wokawoka, 2019-04-02:
Cool, I did ask because since the 4k screen is very energy hungry 10 hours seemed to me a miracle.
I'm going to buy a x1 extreme soon and for this reason I'm probably going for the FHD as well for energy consumption reasons.
Thanks for sharing your findings and for the pop!os suggestion, I'll surely give it a go!

PR #71 (2019-06-11): Improved Hardware support Table
Added models confirmed in issues, 
The ordering is set as: Works, Alphabetical
Made table larger to accept larger device names

Fixes #60 
Fixes #58 
Fixes #57 
Fixes #56
Fixes #55
commit 272f9722 (2019-06-11): Improved Hardware support Table (#71)

* Improved Hardware support Table

Added models confirmed in issues, 
The ordering is set as: Works, Alphabetical
Made table larger to accept larger device names

Fixes #60 
Fixes #58 
Fixes #57 
Fixes #56

* Added model in #55 based on new response

The issue was missing the CPU model. The author just replied adding the info.

Fixes #55

* removed capital p from ThinkPad
closed 2019-06-11 by georgewhewell