georgewhewell/undervolt #61: Gui

zmalrobot, 2019-04-20:
Hi! 
I'm new here but i made a small interface gui for your script, may i publish it?

[image]

**Feature:** 
1) Applying profile
3) Reset all to 0
3) Load and save profile
4) Stess cpu after applying voltage

**Missing:**
1) Temperature target, on my pc don't work so i have no way to test it...
2) A lot of testing... i've made it for my personal use, so it will not cover all cases 

**Ps:**
it i's made in java and it will not include your script but there is a script that download it from here ( i'm a lazy guy ;-) ) and i've mentioned you in the faq and in the source code (...obviously i'll upload that too ) !

Kronos256, 2019-04-21:
i like it but what is uncore and analogio ?

georgewhewell, 2019-05-03:
Nice- looks awesome! If you want you can open a PR adding link to your project to README :)

closed 2019-05-03 by georgewhewell