georgewhewell/undervolt #62: What   CACHE  UNCORE  ANALOGIO options actually do?

ksuhiyp, 2019-04-29:
Are they required, should they be as same as core offset, I am sticked to core cache gpu for  110 110 75 respectively 97 emp , on  8550U Arch 5.07, anybody reached higher limits?

georgewhewell, 2019-05-03:
Don't know, sorry! They are not required to be set, and can be different to other offsets.

I haven't noticed any difference in temp/power when setting them on my platform (Kaby lake). Maybe they have effect on some other platforms

closed 2019-05-03 by georgewhewell

StefanFabian, 2019-05-04:
I've had the same question and googled a bit. It might be wrong but I found that uncore is the bus, RAM etc. and you shouldn't underclock AnalogIO because it doesn't have an advantage but may cause bad stuff to happen.