georgewhewell/undervolt #63: OSError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

ShakirKhurshid, 2019-05-02:
__

ShakirKhurshid, 2019-05-02:
When I use "**undervolt -r**"
it outputs the error `Traceback (most recent call last):
  File "/home/user/.local/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/home/user/.local/lib/python2.7/site-packages/undervolt.py", line 271, in main
    tjunc=read_temperature(),
  File "/home/user/.local/lib/python2.7/site-packages/undervolt.py", line 157, in read_temperature
    return (read_msr(0x1a2) & (127 << 24)) >> 24
  File "/home/user/.local/lib/python2.7/site-packages/undervolt.py", line 54, in read_msr
    f = os.open(n, os.O_RDONLY)
OSError: [Errno 13] Permission denied: '/dev/cpu/0/msr'
`
even after  disabling secure boot, it still shows me this temprature

retromuz, 2019-05-02:
Use
$ sudo undervolt -r

ShakirKhurshid, 2019-05-03:
tried, it shows unknown command 'undervolt'

georgewhewell, 2019-05-03:
`sudo $(which undervolt)`

ShakirKhurshid, 2019-05-04:
The error still remains, is there a possibility that I can't use it because of hardware restrictions
I have an acer predator laptop with an i5 8300H,

StefanFabian, 2019-05-04:
No, the issue is that you didn't install it to a global path.
Maybe try `pip uninstall undervolt` and `sudo -H pip install undervolt`.

auyer, 2019-06-10:
This issues should be closed.

Should we update the docs to point the need of root permissions to run this tool ?

ShakirKhurshid, 2019-06-11:
These solutions didn't work, maybe I think it could be a hardware thing.

auyer, 2019-06-11:
@ShakirKhurshid  Were you able to run undervolt as a super user  (aka sudo) ?

ShakirKhurshid, 2019-06-12:
@auyer NO, didnt worked.

georgewhewell, 2019-06-12:
It's strange- usually If you still get `permission denied` even running with root it means kernel lockdown is enabled, same as #16- but you get the error when writing to the MSR not reading- your traceback has it during read. Can you post your distro/version/kernel @ShakirKhurshid? It would be nice if we had some way of reading whether lockdown is enabled or not, but last time I checked there was no way.

auyer, 2019-06-12:
@ShakirKhurshid by running I meant getting the script to load instead of getting `unknown command 'undervolt' `, not necessarily getting it to work.
It is not yet clear if you were able to run it.

ChrisjanWust, 2019-12-15:
Here's some more detail on the issue:

Running `undervolt --read` gives the following to me, similar to what @ShakirKhurshid posted in the original issue description:

```
Traceback (most recent call last):
  File "/home/user1/.pyenv/versions/3.7.5/bin/undervolt", line 11, in <module>
    load_entry_point('undervolt==0.2.10', 'console_scripts', 'undervolt')()
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 280, in main
    tjunc=read_temperature(),
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 165, in read_temperature
    return (read_msr(0x1a2) & (127 << 24)) >> 24
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 62, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'
```

`sudo undervolt -r` does not find the command:

```
sudo: undervolt: command not found
```

However, `sudo $(which undervolt)` runs fine (as does just `undervolt`), but `sudo $(which undervolt) -r` throws another, but different permission error:

```
temperature target: -20 (80C)
Traceback (most recent call last):
  File "/home/user1/.pyenv/versions/3.7.5/bin/undervolt", line 11, in <module>
    load_entry_point('undervolt==0.2.10', 'console_scripts', 'undervolt')()
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 284, in main
    voltage = read_offset(plane)
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 178, in read_offset
    write_msr(value_to_write)
  File "/home/user1/.pyenv/versions/3.7.5/lib/python3.7/site-packages/undervolt.py", line 53, in write_msr
    os.write(f, pack('Q', val))
PermissionError: [Errno 1] Operation not permitted
```

Environment details:
Running `Python 3.7` through `pyenv` on `Ubuntu 19.10`. I'm wondering whether the issue is related to the way Python is installed. `sudo pip` throws a `command not found` error.

StefanFabian, 2019-12-15:
`sudo undervolt` does not find the command because it's not in the global path.
I assume running `which undervolt` returns something like `$HOME/.local/bin/undervolt`.
`sudo $(which undervolt)` runs fine because `which undervolt` is first evaluated as your user (because of the `$()`) and returns the full path to the script, which is then called explicitly as root user.
Regarding the other error, it seems like your msr is readonly even for root (unlikely) or `pyenv` somehow spawns a subprocess with different permissions but I would suggest opening a new issue with more details such as CPU model, Laptop model etc.
Can you try running it directly with python without `pyenv`?

georgewhewell, 2019-12-15:
Seems like kernel lockdown

conraddd, 2019-12-21:
```
sudo undervolt -r
temperature target: -8 (92C)
Traceback (most recent call last):
  File "/usr/local/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/undervolt.py", line 298, in main
    voltage = read_offset(plane)
  File "/usr/local/lib/python2.7/dist-packages/undervolt.py", line 192, in read_offset
    write_msr(value_to_write)
  File "/usr/local/lib/python2.7/dist-packages/undervolt.py", line 67, in write_msr
    os.write(f, pack('Q', val))
OSError: [Errno 1] Operation not permitted
```

conraddd, 2019-12-30:
I solved the issue by disabling security boot from bios

hunt0x3r, 2020-05-08:
Having a similar issue. On `sudo undervolt` I get: `sudo: undervolt: command not found`. `sudo $(which undervolt)` returns:
```
Traceback (most recent call last):
  File "/home/hunter/.local/bin/undervolt", line 6, in <module>
    from undervolt import main
ModuleNotFoundError: No module named 'undervolt'

```
This makes me think this is a python version problem on my end? I have both 2.7 and 3.7 installed. Maybe the script is calling 2.7 when it should be calling 3.7?

hunt0x3r, 2020-05-08:
Not a version control problem. Script calls proper version.

lllldronellll, 2020-06-04:
I'm also having very similar results using Pop OS 20.04

Traceback (most recent call last):
  File "/usr/local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 439, in main
    temp = read_temperature(msr)
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 191, in read_temperature
    return (read_msr(msr.addr_temp) & (127 << 24)) >> 24
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 88, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

closed 2020-06-08 by georgewhewell

ShakirKhurshid, 2020-06-19:
use `sudo pip install undervolt`  instead of `pip install undervolt`
then run it with root access
`sudo undervolt --core -100 --cache -100`
This should do the trick for you !

AliTajelsir, 2020-07-02:
I don't know if this is going to be helpful or not but:
- I always install `pip` packages by `pip install --user <package>`, so I'd not mess with my system packages. When I installed `undervolt` using the same method and I ran `sudo undervolt r` I had the before mentioned bug:
```
Traceback (most recent call last):
  File "/home/ali/.local/bin/undervolt", line 5, in <module>
    from undervolt import main
ModuleNotFoundError: No module named 'undervolt'
```
- Elevating the command while preserving my environment solved the problem for me `sudo -E undervolt -r`.

faedy2, 2021-11-08:
I also have this error
I was able to install `sudo pip install undervolt`, but would I be able to have the adjustments run at boot with sudo permissions?

StefanFabian, 2021-11-09:
Yes if you are using systemd (if you're on linux and don't know, you probably are) you can follow the instructions in the readme to set up a service to run the adjustments.
Make sure your adjustments are stable before creating a service that applies them on boot!

PR #113 (2020-06-08): Check if the script is run by the root where necessary
Fixes #110 
Fixes #63
When I run `undervolt.py  --read` and `undervolt.py --cache -100` 
The errors given by the script as follows:
```
Traceback (most recent call last):
  File "./undervolt.py", line 470, in <module>
    main()
  File "./undervolt.py", line 449, in main
    temp = read_temperature(msr)
  File "./undervolt.py", line 195, in read_temperature
    return (read_msr(msr.addr_temp) & (127 << 24)) >> 24
  File "./undervolt.py", line 92, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/d [... cut]