georgewhewell/undervolt #66: Intel gpu not undervolting

gioiann, 2019-05-08:
I've tried undervolting a dell xps 15 9560 with an i7-7700HQ
The cpu undervolting is working cause by using -150mv or less the pc freezes.
The gpu undervolting however seems like to have no effect at all since i can set -500mv and everything works fine. By setting more than a certain value like -2000mv and then reading the settings -2000 becomes positive, it doesn't show -2000 in the reading but 2000 without -

I'm sure i'm not using the nvidia gpu.

georgewhewell, 2019-05-11:
Strange- this has been reported before but with different CPU: https://github.com/georgewhewell/undervolt/issues/43

gioiann, 2019-05-11:
I just noticed that undervolting values aren't reset when the pc is restarted. So for example I can undervolt on windows 10 using Intel Extreme Tuning Utility and reboot in ubuntu and the undervolt is still applied and i can also read the offset values. The same happens from ubuntu to windows 10.

So i tried setting everything to -130mv and the gpu to -300 on ubuntu then by rebooting into windows these are the offsets XTU reads

[image]

**Processor Graphics Voltage Offset** is set to -300mv but **Processor Graphics Media Voltage Offset** is set to -130mv

So, the linux undervolt is correctly setting the "Voltage Offset" but not the "Media Voltage Offset"

And there is no way these 2 offsets can be different in XTU.