georgewhewell/undervolt #85: Advantages over intel-undervolt

lhk, 2019-10-21:
The Readme presents this project as a thin wrapper on top of Intel-undervolt. Apart from small differences in the cli, are there reasons to use this project over the other? Are there differences in the featureset?

georgewhewell, 2019-10-21:
The project referenced (https://github.com/mihic/linux-intel-undervolt) is purely informational- there is no tool or CLI.

lhk, 2019-10-21:
Oh sorry, my bad. I had found this at the same time as your project : https://github.com/kitsunyan/intel-undervolt

Since the naming is so similar, I thought you were linking to it.

So I assume the two projects (undervolt and intel-undervolt) are pretty much in direct competition? Two implementations of the same functionality?

georgewhewell, 2019-10-21:
Yes they do pretty much same thing. The project you linked seems to have more features but requires compiling and bit more complex to distribute

closed 2019-10-21 by georgewhewell