georgewhewell/undervolt #86: Ramifications of running this every 30 seconds?

Ph0rkoz, 2019-10-24:
Do we really want undervolt to run every 30 seconds? Would this not produce wakeups, etc?

Is there a better length of time to re-set the undervolt?

Ph0rkoz, 2019-10-27:
Ok, found one ramification.... undervolt spams the hell out of the logs.

retromuz, 2019-10-29:
That's true about the logs.
However, my idle consumption stays well below 10W (~ 6 - 7). Sometimes it even goes down to ~ 4 - 5 Watts. I comfortably got more than 10 hours of battery life. Time to time I run ~2 virtual machines under VirtualBox. Do remote desktop via Remmina all the time. Still get at least 8 hours of battery. This is pretty good for me.

Ph0rkoz, 2019-10-31:
Yes, undervolting itself = good. Spamming my logs = killing my ssd.

georgewhewell, 2019-11-01:
dupe of #45 

if you don't want any output then drop the `-v` flag from your service

closed 2019-11-01 by georgewhewell

Ph0rkoz, 2019-11-03:
Thank you. It really is that easy. Please mention it in the docs since it wasn't as obvious as it should have been.