georgewhewell/undervolt #90: Works on Asus Vivobook prto 15 n580gd

sangallidavide, 2019-11-29:
Undervolting works fine on my laptop.
Readout of voltage done via i7z

I did 
`sudo $(which undervolt) --core -150 --cache -150`
and I do not experience problems. It was also ok with -175 on both.
Instead when I tried to lower even more or to do 
`sudo $(which undervolt) --core -100 --cache -100 --uncore -100 --analogio -100`
it became unstable. I do not now which of the two is the source (analogio or uncore)

sangallidavide, 2019-12-03:
I add that instead
```sudo $(which undervolt) --temp 85```
does not work
both before and after the read command gives
```
sudo $(which undervolt) -r
temperature target: -4 (96C)
core: -150.39 mV
gpu: 0.0 mV
cache: -150.39 mV
uncore: 0.0 mV
analogio: 0.0 mV
```
If I try
```sudo $(which undervolt) --temp -10```
it returns erro

georgewhewell, 2019-12-15:
@sangallidavide interesting, maybe something in your BIOS prevent changing temperature offset. AFAIK adjusting uncore and analogio do not really affect power consumption, they are just included for completeness.

closed 2019-12-15 by georgewhewell