georgewhewell/undervolt #91: ERROR:root:Failed to apply core: set -0.9765625, read 0.0

przemyslawmaluj, 2019-12-03:
I used undervolt without any issues for several months. Last week I noticed that something is not right. Then I ran `sudo undervolt -v --core -1 --cache -1
` and got 

> INFO:root:Setting core offset to -1mV
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/0/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/1/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/2/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/3/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/4/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/5/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/6/msr
INFO:root:Writing 0x80000011ffe00000L to /dev/cpu/7/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/0/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/1/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/2/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/3/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/4/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/5/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/6/msr
INFO:root:Writing 0x8000001000000000L to /dev/cpu/7/msr
INFO:root:Read 0x0 from /dev/cpu/0/msr
ERROR:root:Failed to apply core: set -0.9765625, read 0.0

Secure Boot is disabled. I have i7-8650U, undervolt 0.2.10. My system was running fine with --core -150 --cache --150

georgewhewell, 2019-12-10:
Hm, strange.. no ideas.. did anything else change since it was working? bios update?

gierdo, 2019-12-15:
I've got the same issue, and I also suspect a bios or microcode (if that can affect it, independently from the bios?) update.

przemyslawmaluj, 2019-12-16:
Right, it looks like BIOS was updated.

gierdo, 2019-12-18:
Just for the fun of it I tested the behavior with https://github.com/kitsunyan/intel-undervolt
Same symptoms, of course.
I guess we have to wait for / hope for a new bios that may or may not allow us to undervolt again.
But with plundervolt, I'm afraid the exact opposite is going to happen for many of us!

przemyslawmaluj, 2019-12-18:
That sucks... CPU without undervolting is heavily throttling almost all the time.

przemyslawmaluj, 2020-01-03:
I rolled back BIOS to previous version and can use undervolt again.

georgewhewell, 2020-01-04:
Thanks for confirm.

closed 2020-01-04 by georgewhewell

owenjm, 2020-04-02:
I'm also suddenly experiencing this issue (been using undervolt for months, and is suddenly not working) ... and downgrading the bios does not fix it.  (The downgrade worked fine except for updating the Intel ME firmware -- which I don't think should affect this, but I'm not entirely sure?)   Any suggestions here would be greatly appreciated, as undervolting massively helps in preventing my laptop's fans from running constantly.

owenjm, 2020-04-03:
(Sorry, scratch that -- needed to reset BIOS to defaults, and with the BIOS downgrade it's back.)

Michele2k1, 2020-04-10:
I have got the same error
[image_log(https://imgur.com/a/e6NX5Fc)
on dell xps 15 7590
any way to fix? I just installed undervolt

wlainer, 2020-05-30:
Its' no necessary downgrade bios, I have just reseted bios to factory's default and it worked.

Kmiller0313, 2020-07-10:
> Its' no necessary downgrade bios, I have just reseted bios to factory's default and it worked.

Just resetting the bios to factory default and it still doesn't work.

TerrorSquad, 2020-07-27:
Same here, cannot undervolt.

❯ sudo undervolt --core -100 --cache -100 
ERROR:root:Failed to apply core: set -99.609375, read 0.0

Dell XPS 15 9570
Linux xps-9750 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
I was running the most recent BIOS update and reverted to 1.4.1 (cannot revert to an earlier version), and even after the revert undervolting is not possible.

What's weird is, undervolting is possible under Windows using Throttlestop. I set the the voltage offset to -125mV for core and cache and temps went nearly 10C degrees. A substantial improvment.

I was reading online how kernel lockdown is actually responsible for this, denying access to /dev/mem paths required for undervolting. Also while reading on lots of places it said that disabling Secure Boot would resolve the issue. Secure Boot is disabled but undervolting under Linux is still not possible.

Can we get an update why is this happening and/or how to work around the issue and fix undervolt?

afrish, 2020-08-01:
This solution worked for me: https://www.dell.com/community/XPS/Intel-Extreme-Tuning-Utility-Undervolt-Fails-after-Bios-update/td-p/7542653. Basically, go to BIOS -> Restore Settings -> Factory Settings, and after that **undervolt** works

moritzschaefer, 2020-08-04:
@afrish thanks for the workaround. I'm honestly too afraid to crash my Linux-installation when restoring "factory setting". I needed to change quite some settings in the Bios before installation of NixOS was possible..

afrish, 2020-08-04:
@moritzschaefer I'm using Kubuntu 20.04 on Dell XPS 9570 with Intel Core i5 and no discrete video card. After resetting to the default settings I just booted as usual and everything worked great. One small thing was changed: top-level F buttons changed their mode, but it's very easy to change them back. There are not so many settings in BIOS, maybe you can make couple photos of your BIOS screen before resetting?

TerrorSquad, 2020-08-04:
@afrish thanks for the solution. It worked for me. No settings were lost after the reset. Also, after booting back into Windows, BIOS update started by itself, I presume triggered when Windows started to boot. BIOS is at the latest version now - 1.16.2 and **undervolt works** :tada:

moritzschaefer, 2020-08-04:
Thank you for the encouragement (that's what I actually nedded). I'll try it later..

moritzschaefer, 2020-09-21:
The issue persists for me! I actually have the BIOS version 1.19.2, maybe that's the issue. I'm therefore anyways surprised that @TerrorSquad reports version 1.16.2 to be the most recent one..

sangemaru, 2020-11-28:
Same issue on XPS 15 9560 BIOS 1.21. Tried bios defaults, factory reset, etc. Probably not an issue with undervolt, as ThrottleStop won't work either.

notstarboard, 2024-04-16:
This error has prevented me from undervolting my lab computer (details [here](https://askubuntu.com/questions/1510600/undervolting-help-is-it-possible-to-roll-back-an-apple-bios-to-an-older-versio). Any help would be much appreciated!

By the way, are there any other working undervolt tools on Linux? Rolling back to an older BIOS version is only a possible workaround for folks who had an older BIOS version to begin with. People have been hitting this error with undervolt since 2019, so it stands to reason that there are a lot of people out there with newer machines they're not able to undervolt. Are they just stuck, or is there some other working software out there?

GermanAizek, 2024-10-09:
Same error on server motherboard Lenovo RD450X

luckylinux, 2025-01-15:
Same error on a Supermicro X11SSL-F Motherboard and Intel Xeon E3-1240 v5 with BIOS Version 2.6 (quite Old).

Seems to work on a Supermicro X10SLL-F Motherboard and Intel Xeon E3-1275 v3 with BIOS Version 3.4.

Funnily enough it seems to also work on an ASUS P9DWS and Intel Xeon E3-1245 v3 :confused: (where I do NOT need it since I can tweak the Voltage in BIOS already :rofl).

Does anybody know why Resetting to Factor Defaults seems to work ?

I can only see this when inspecting the "disassembled" BIOS using UEFITool + ifrextractor(my [Procedure](https://github.com/luckylinux/aspm-troubleshooting?tab=readme-ov-file#patching-bios-settings)):

```
OneOf Prompt: "    Platform Power Limit Lock", Help: "When enabled, PLATFORM_POWER_LIMIT MSR will be locked and a reset will be required to unlock the register.", QuestionFlags: 0x10, QuestionId: 0x64, VarStoreId: 0x1, VarOffset: 0x8F, Flags: 0x10, Size: 8, Min: 0x0, Max: 0x0, Step: 0x0
						OneOfOption Option: "Disabled" Value: 0
						OneOfOption Option: "Enabled" Value: 1, Default, MfgDefault
					End 
```

and

```
OneOf Prompt: "Overclocking lock", Help: "FLEX_RATIO(194) MSR", QuestionFlags: 0x10, QuestionId: 0x57, VarStoreId: 0x1, VarOffset: 0x4D, Flags: 0x10, Size: 8, Min: 0x0, Max: 0x0, Step: 0x0
					OneOfOption Option: "Disabled" Value: 0, Default, MfgDefault
					OneOfOption Option: "Enabled" Value: 1
				End 

```

So I can only guess that playing with the `Platform Power Limit Lock` BIOS Setting (and set it to `Disabled`) is what actually helps.

The other Setting/Help Menu isn't something I recall having seen in BIOS though.

statikana, 2026-10-18:
Same issue, NixOS Yarara 26.05 on Intel 12700H. Reset BIOS to default, still getting the same error:
```
ERROR:root:Failed to apply core: set -150.390625, read 0.0
```
I did have to change on BIOS setting (secure boot, on -> off), though this is the only thing I changed since otherwise I can't boot. Any Linux wizards figured this one out?