georgewhewell/undervolt #92: Set power limit

StefanFabian, 2019-12-10:
I've made a fork with the added option to set the short term and long term power limit.
Unfortunately, my laptop's firmware overrides my settings so I have no means of testing it.
I can read the limits correctly and setting them also updates the values correctly but they don't affect anything.
If someone could test it and confirm that it actually works, I can make a PR.
Added options are:
```
  -p1 POWER_LIMIT_LONG TIME_WINDOW_LONG, --power-limit-long POWER_LIMIT_LONG TIME_WINDOW_LONG
                        P1 Power Limit (W) and Time Window (s)
  -p2 POWER_LIMIT_SHORT TIME_WINDOW_SHORT, --power-limit-short POWER_LIMIT_SHORT TIME_WINDOW_SHORT
                        P2 Power Limit (W) and Time Window (s)
  --lock-power-limit    Locks the set power limit. Once they are locked, they
                        can not be modified until next RESET (e.g., Reboot).
```

**Disclaimer:** Use at your own risk. I do not give any guarantees nor am I liable for any potential damages.

georgewhewell, 2019-12-10:
Awesome, thanks! Want to make a PR and we can discuss there? I think I can test on my laptop, although have never tried

georgewhewell, 2020-06-08:
Closed via #93

closed 2020-06-08 by georgewhewell