georgewhewell/undervolt #98: Ubuntu rights

hubyhuby, 2020-03-16:
After doing the pip install, the "undervolt --read" command doesn t work on my Ubuntu Mate 18.04 LTS.
I suspect a right permission problem...
Trying with "sudo undervolt --read" says undervolt doesn t exist.

undervolt --read
modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "/home/bob/.local/bin/undervolt", line 11, in <module>
    sys.exit(main())
  File "/home/bob/.local/lib/python2.7/site-packages/undervolt.py", line 254, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1

Teardrop13, 2020-03-16:
Try running pip with sudo. I've had the same error

hubyhuby, 2020-03-16:
Thanks @Teardrop13 
I Have tried the following but i get exactely the same error when doing undervolt --read => modprobe: ERROR: could not insert 'msr': Operation not permitted

sudo -H pip install undervolt
Collecting undervolt
  Downloading https://files.pythonhosted.org/packages/3e/6d/d850608fc9fb5f8ec07c802ebf9a6e30987f814ca81c2f2fdec7996d8afc/undervolt-0.2.11.tar.gz
Building wheels for collected packages: undervolt
  Running setup.py bdist_wheel for undervolt ... done
  Stored in directory: /root/.cache/pip/wheels/b8/e1/9b/1bcc46e08ceba6a5c7401aa5ff13ec23fa0dce755b39a1a92b
Successfully built undervolt
Installing collected packages: undervolt
Successfully installed undervolt-0.2.11

hubyhuby, 2020-03-16:
Ok
Now it works I have to sudo aswell for every command after install :
sudo undervolt --read
temperature target: -20 (80C)
core: 0.0 mV
cache: 0.0 mV
analogio: 0.0 mV
uncore: 0.0 mV
gpu: 0.0 mV

georgewhewell, 2020-03-18:
Working as intended

closed 2020-03-18 by georgewhewell