georgewhewell/undervolt #105: Error, I cannot change voltages

FernandoNobel, 2020-04-21:
Hi everyone!

I have installed undervolt but when I try to execute this command:

`sudo undervolt --core -100 --cache -100`

I get the following error:

`ERROR:root:Failed to apply core: set -99.609375, read 0.0`

And if I read the values:
```
sudo undervolt --read
temperature target: -3 (97C)
core: 0.0 mV
gpu: 0.0 mV
cache: 0.0 mV
uncore: 0.0 mV
analogio: 0.0 mV
```

What does this error mean?

My cpu is a Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz.

aalepis, 2020-04-22:
Same error for Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz. I have secure boot set to off btw.

The temps though seem to have dropped by 10-12 C so maybe the values are written but not read correctly.

przemyslawmaluj, 2020-04-22:
Please look at my issue https://github.com/georgewhewell/undervolt/issues/91 - exact same problem. Try downgrading BIOS.

marcomarmelade, 2020-04-24:
@przemyslawmaluj I have the same issue on my xps 7590. I also tried downgrading BIOS as suggested (1.6 --> 1.5). 
Just for reference which version of BIOS fixed the issue for you?

marcomarmelade, 2020-04-26:
so I was able to fix it for my machine. Just in case any one else is having the same issue here is what worked for me: down grade bios plus enter bios option and find “reset to defaults”, then make sure to choose “reset to factory defaults”

mich2k, 2020-04-29:
@marcomarmelade which version?
I ve XPS 15 7590

marcomarmelade, 2020-05-04:
@mich2k I have the XPS 15 7590 with the i7-9750H CPU. Did it worked for you?

mich2k, 2020-05-04:
@marcomarmelade 
I have not tried it, since I would not like to downgrade bios to make this work (I have your same CPU)

marcomarmelade, 2020-05-04:
I can definitely recommend it. Works super stable on my machine. Together with i8kmon my fans dont come on nearly as often as before. Thus i can enjoy working on it much more. And the downgrade of the bios didn't effect me at all.

mich2k, 2020-05-04:
Which version have you downgraded to?
(Are you it?)
Anyway with tlp+thermald+powertop 
I went from 1 hour to 4-5 hours 
If you suggest me to downgrade I would try too

marcomarmelade, 2020-05-04:
I downgraded it to 1.5. But you should definitely decide this for your own, since its your machine. Only thing I can tell you is that I have right now 5 hours left with 81% battery and 4k display running. Hope this helps in making a decision.

mich2k, 2020-05-04:
I got 4k oled too
Can I ask you what parameters (core, cache etc) have you chose?
(Yes I know that these are different from each machine)

mich2k, 2020-05-05:
ok I can confirm than on DELL XPS 15 7590, @ i7-9750H is working, so the bios 1.6.0 is unsupported
I had to restore bios (even if I had to untick secure boot and remember to disable RAID) because otherwise this still did not work
thank you very much, I will check if this has benefits and how much

mich2k, 2020-05-31:
Edit: just updated to BIOS 1.7.0
Still working

marcomarmelade, 2020-06-01:
thanks for the notification. 1.7 is also working on my system, so than there was only the issue with the 1.6 version

afrish, 2020-08-04:
I posted it in another linked issue, will cross-post it also here.

This solution worked for me: https://www.dell.com/community/XPS/Intel-Extreme-Tuning-Utility-Undervolt-Fails-after-Bios-update/td-p/7542653. Basically, go to BIOS -> Restore Settings -> Factory Settings, and after that undervolt works.

I'm running undervolt on Dell XPS 9570 with Kubuntu 20.04, Intel Core i5 and no discrete video card. I had no problem booting after the reset and did no changes to BIOS settings after that

driver-by, 2020-10-30:
If you have the same error on MSI laptop (e.g. GS75 Stealth 10SFS), try this out:

- Enable hidden features in BIOS (Ctrl Right + Shift Right + Alt Left + F2)
- Go to "Advanded" -> "OverClocking Performance Menu"
- Enable "OverClocking Feature"
- Enable "XTU Interface"
- Save & Exit