georgewhewell/undervolt #107: support for i7-9750H

Arthav24, 2020-04-25:
Successfully undervolted my i7-9750H but the **systemctl status undervolt** display this
```
 undervolt.service - undervolt
   Loaded: loaded (/etc/systemd/system/undervolt.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2020-04-25 20:05:32 IST; 2h 39min ago
  Process: 1003 ExecStart=/home/arthav/undervolt/undervolt.py -v --core -125 --cache -125 (code=exit
 Main PID: 1003 (code=exited, status=0/SUCCESS)

Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/4/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/5/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/6/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/7/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/8/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/9/msr
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/10/ms
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/11/ms
Apr 25 20:05:32 arthav-G7 undervolt.py[1003]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
Apr 25 20:05:32 arthav-G7 systemd[1]: Started undervolt.
```
 So is undervolt writting to MSRs of CPU 4 to 11 only? and Then wy reading only from CPU 0 ??

Is this natural behaviour or something abnormal please look into! let me know if you want any debug files and info. 
I validated the undervolt of -125mV in Corefreq the max VCore in load drops immediately on applying undervolt.

retromuz, 2020-04-26:
Increase the height of your terminal window that you run command on.

retromuz, 2020-04-26:
That is exactly what I see in mine if I run for a consecutive time. Do a "sudo journalctl -f" and wait for the "undervolt.timer" to kick in. You'll see the full output.

Arthav24, 2020-04-26:
Thanks @retromuz 
`Apr 26 11:57:11 arthav-G7 systemd[1]: Starting undervolt...
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Setting core offset to -125mV
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000011f0000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000001000000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Setting cache offset to -125mV
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x80000211f0000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/1/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/2/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/3/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/4/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/5/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/6/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/7/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/8/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/9/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/10/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Writing 0x8000021000000000 to /dev/cpu/11/msr
Apr 26 11:57:11 arthav-G7 undervolt.py[11515]: INFO:root:Read 0xf0000000 from /dev/cpu/0/msr
Apr 26 11:57:11 arthav-G7 systemd[1]: Started undervolt.
`
@georgewhewell Please add Dell G7 7590  i7-9750H in the supported list then. 
Also undervolting works till bios version 1.11

viocost, 2020-04-26:
Same issue. Updated bios to 1.12 - undervolt stopped working. 
My config: Dell G5 5590 core i7-9750H
Bios version 1.11.1 is still working. I had to downgrade and reset bios settings to factory defaults after downgrading in order for undervolt to work. Downgrading only didn't have any useful effect.

closed 2022-06-11 by Arthav24