georgewhewell/undervolt #108: Is there a "default" undervolt applied when you first install?

spiff72, 2020-04-29:
I just installed Ubuntu 20.04 as a fresh install (after using 19.10 on my Thinkpad X1 Extreme Gen 2), and I have Throttlestop installed on the Windows side of my dual-boot.

I copied the throttlestop.ini file from my windows setup to a folder on my Ubuntu install, but it is buried in a folder and I did nothing but copy the file over.

After first installing undervolt with pip3, I ran sudo undervolt --read, and it gives me values that are very similar to those I use in Throttlestop.  Is this just coincidence?  Or does the setting from Throttlestop in windows somehow get carried over to the linux side of my dual boot?

I get these values when running the --read option:
temperature target: -20 (80C)
core: -125.0 mV
gpu: -75.2 mV
cache: -125.0 mV
uncore: 0.0 mV
analogio: 0.0 mV

Thanks!

spiff72, 2020-04-29:
Well this is really weird - I just ran the command again, and now it states:
temperature target: -20 (80C)
core: 0.0 mV
gpu: 0.0 mV
cache: 0.0 mV
uncore: 0.0 mV
analogio: 0.0 mV

I have no idea how the other values above were generated...

georgewhewell, 2020-05-09:
some users have reported their settings persist through reboots, so I think it is likely the first set of values you got persisted from throttlestop in windows

closed 2020-05-09 by georgewhewell