georgewhewell/undervolt #109: Undervolt not set for all cores

fiorano10, 2020-05-01:
I have a system service to run after 30 seconds of reboot

`Service:
[Unit]
Description=Set CPU undervolt
After=suspend.target
After=hibernate.target
After=hybrid-sleep.target

[Service]
Type=oneshot
ExecStartPre=/bin/sleep 30
ExecStart=/usr/local/bin/undervolt -v --core -150 --cache -150 --gpu -250 --temp 70

[Install]
WantedBy=multi-user.target
WantedBy=suspend.target
WantedBy=hibernate.target
WantedBy=hybrid-sleep.target`

`INFO:root:Writing 0x1e000000 to /dev/cpu/3/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/4/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/5/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/6/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/7/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/8/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/9/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/10/msr`
`INFO:root:Writing 0x1e000000 to /dev/cpu/11/msr`

Not sure why it doesn't set the undervolt for all cores, it does so if I enter the command from the terminal.

georgewhewell, 2020-05-09:
How are you getting the log? `systemctl status undervolt` will by default only show the last few lines of output. Can check `journalctl -xefu undervolt` and see if you see whole output there

fiorano10, 2020-05-11:
Checked `journalctl`, it does undervolt all cores.

closed 2020-05-11 by fiorano10