georgewhewell/undervolt #114: Dell XPS 15 9570	i9-8950HK	Not OK

phackwer, 2020-05-21:
Hi,

There was an update on Bios and since then the undervolt stopped to work. Any better suggestions than "downgrading" bios?

Thanks

retromuz, 2020-05-21:
journalctl or dmesg log would help.

eadwu, 2020-05-22:
Seems like it's blocked now.
`ERROR:root:Failed to apply core: set -99.609375, read 0.0`

bavay, 2020-05-25:
Same for me since updating to bios 1.6.0 on my XPS 15 7590 i7-9750H... I've complained to Dell that they should offer a way to undervolt directly from the bios and now it is not even possible from the OS anymore.

worldofgeese, 2020-06-04:
I am experiencing the same error on my XPS 13 9380 running BIOS version 1.10.0.

rtfpessoa, 2020-06-11:
I can confirm @worldofgeese solution working fine. Weird tough.

bavay, 2020-06-19:
@worldofgeese , @rtfpessoa For the sake of clarity: when doing a factory reset of the BIOS, does it means that it reverses to an older version of the BIOS or that you still have the latest bios update but without your settings?

rtfpessoa, 2020-06-20:
@bavay I think it only reverted the settings

prismspecs, 2020-06-20:
BIOS reset did not work for me XPS 7590

bavay, 2020-06-24:
On my XPS 7590, "reset to factory settings" in BIOS did the trick. The most important settings for my Linux installation were not even lost (for example AHCI mode, no secure boot, while some of my other, less important settings got reset) and now undervolt works again. For information, here are my undervolt settings for this specific model:
undervolt --core -125 --cache -125 --uncore -125 --analogio -125 --gpu -125 --temp 95

(more aggressive undervolting was not stable).

prismspecs, 2020-07-04:
I do not have a "reset to factory settings" option, what BIOS version are you on?

Dedas, 2020-07-05:
BIOS reset did not work for me. Dell XP 15 9560

mikehardy, 2020-07-10:
@prismspecs it was an option for me in the dialog that popped up when I clicked one of the buttons at the bottom of every screen, next to "apply" I believe the label was "reset", then you can choose how, factory default was an option. My experience was similar to @bavay in that none of the settings that were important to running linux were altered, but as with all things BIOS-related, take great care :-), good luck

luispabon, 2020-07-14:
@Dedas you also need to reset your BIOS settings after downgrading to v1.18.0

Dedas, 2020-07-15:
> @Dedas you also need to reset your BIOS settings after downgrading to v1.18.0

Thanks! Yes, after I did a downgrade and reset the bios it worked.

amits1995, 2020-07-16:
Confirmed - I have a Dell XPS 9560, I downgraded the BIOS back to version 1.18.0 using the bios installation file in dells website.  
After that, Factory reset the BIOS settings and undervolt works again.

luispabon, 2020-07-16:
For someone else who needs to downgrade the firmware, you can use the fwupgmgr tool (not sure about the exact CLI args for it) or `gnome-firmware`:

[image]
[image]

ManifoldFR, 2020-12-05:
I'm on a Dell XPS 7590 with firmware version 1.8.1. Resetting to factory defaults in BIOS did the trick -- secure boot was stayed disabled and AHCI was kept on, nothing else major changed. I'm now able to adjust the voltage. Thanks @worldofgeese !