georgewhewell/undervolt #120: Undervolting uncore does not work

SanZamoyski, 2020-06-21:
Hi!

I'm on Dell Latitude E7440 with Intel(R) Core(TM) i5-4300U CPU @ 1.90GHz.
I have successfully set core and cache to -85 (-90 cause instability), but have problem with uncore:

```
sudo undervolt --uncore -10 -v
INFO:root:Setting uncore offset to -10mV
INFO:root:Writing 0x80000311fec00000 to /dev/cpu/0/msr
INFO:root:Writing 0x80000311fec00000 to /dev/cpu/1/msr
INFO:root:Writing 0x80000311fec00000 to /dev/cpu/2/msr
INFO:root:Writing 0x80000311fec00000 to /dev/cpu/3/msr
INFO:root:Writing 0x8000031000000000 to /dev/cpu/0/msr
INFO:root:Writing 0x8000031000000000 to /dev/cpu/1/msr
INFO:root:Writing 0x8000031000000000 to /dev/cpu/2/msr
INFO:root:Writing 0x8000031000000000 to /dev/cpu/3/msr
INFO:root:Read 0x30000000000 from /dev/cpu/0/msr
ERROR:root:Failed to apply uncore: set -9.765625, read 0.0
```

Best regards!

teggun1, 2020-06-23:
This happens to me too, using an Acer Aspire V3-572G with an Intel(R) Core(TM) i7-4510U CPU @ 2.00GHz, however, in my case not only does it not work with _uncore_ but with _analogio_ as well.

BentHaase, 2020-07-06:
Same here for an `i7-4600U` on a Lenovo Thinkpad T440s

tvld, 2020-07-12:
After upgrade to Ubuntu 20.04 same issue. It used to work before... although I am not 100% sure when exactly it stopped working...

luispabon, 2020-07-14:
Same problem as @tvld on the same laptop. Last it worked was in 19.10 with the 5.3 kernel.

```
~ sudo undervolt --core -150 --cache -150 --gpu -100 --verbose 
INFO:root:Setting core offset to -150mV
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/0/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/1/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/2/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/3/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/4/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/5/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/6/msr
INFO:root:Writing 0x80000011ecc00000 to /dev/cpu/7/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/0/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/1/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/2/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/3/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/4/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/5/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/6/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/7/msr
INFO:root:Read 0x0 from /dev/cpu/0/msr
ERROR:root:Failed to apply core: set -150.390625, read 0.0
```

luispabon, 2020-07-14:
Ah:

http://forum.notebookreview.com/threads/black-list-adjustable-voltage-control-turbo-ratio-limits-are-locked-out-due-latest-win-update-bios.831450/

I did upgrade to the newest BIOS recently. Gonna try roll back.

luispabon, 2020-07-14:
Actually it didn't require a rollback, but restoring BIOS to factory settings. Stuff works again :+1:

closed 2020-12-27 by georgewhewell

tvld, 2020-12-28:
How would this be resolved? Downgrading is not a resolve!

luispabon, 2020-12-28:
The undervolt tool can't do anything if the system does not allow undervolting. You can blame Dell for not providing an up to date BIOS that addresses Plundervolt without disabling undervolting altogether.

tvld, 2020-12-28:
But it seems that it still works on Windows... right ? Or am I mistaken ?... ))

luispabon, 2020-12-28:
In my case, it didn't. If the firmware doesn't allow you to it doesn't
matter which OS you're running.

georgewhewell, 2020-12-28:
does it work in windows in that you can crash the machine by setting it too low? your log says that setting the offset did not raise any error, but it was not read back correctly. its likely that your windows tool does not try reading back values to see if they were applied (or does not raise an error for these uncore/analogio values)

luispabon, 2020-12-28:
I noticed undervolting wasn't working at all when I was playing a game in Windows. Thermal throttling without makes it (Destiny 2) completely unplayable. Undervolting doesn't make a lot of difference at the usual loads I run while working in Linux, but reduces cpu temps a good 10C at full load in my particular use case.