georgewhewell/undervolt #124: Dell XPS 13 7390 not working

Maverobot, 2020-08-12:
I am not sure whether it is supposed to work with XPS 13 7390. Here is what I got trying to read the stats:

```sh
$ sudo undervolt --read
temperature target: -0 (100C)
core: 0.0 mV
cache: 0.0 mV
analogio: 0.0 mV
uncore: 0.0 mV
gpu: 0.0 mV
powerlimit: 64.0W (short: 0.00244140625s - disabled) / 64.0W (long: 24.0s - enabled)
```

mensfeld, 2020-09-20:
What Bios version do you have?

moritzschaefer, 2020-09-21:
same problem for XPS 15 6560

```
[I] moritz@moxps ~> sudo undervolt --read
[sudo] password for moritz:
temperature target: -3 (97C)
core: 0.0 mV
gpu: 0.0 mV
cache: 0.0 mV
uncore: 0.0 mV
analogio: 0.0 mV
powerlimit: 56.0W (short: 0.00244140625s - enabled) / 45.0W (long: 28.0s - enabled)

```

mensfeld, 2020-09-21:
@moritzschaefer same question as above ;)

moritzschaefer, 2020-09-21:
Sorry @mensfeld :) 

1.19.2

Resetting BIOS to factory setting didn't help unfortunately.

mensfeld, 2020-09-21:
Hmmm, then I have no idea :( Worked for me after resetting bios to factory settings though had to reset twice and after had several CPU failures that had to mitigate by blocking C levels in BIOS.

almasen, 2020-10-11:
Just found this post after finally having fixed undervolting on our XPS 15 9570 & 7590 devices.

From our research, a BIOS update which "fixes" the below vulnerabilities disabled undervolting on many Dell devices. 

> -Firmware updates to address security advisory INTEL-SA-00307 (CVE-2019-14598)
> -Firmware updates to address the Intel Security Advisories INTEL-SA-00289 (CVE-2019-11157) and INTEL-SA-00317 (CVE-2019-14607).

As per the above, I believe @moritzschaefer, you could try going back back to **1.18.0**.
Similarly, on the 7390, @Maverobot, you could try **1.4.0**.

Note, that it is quite unfortunate that you need to roll back your firmware in order to be able to undervolt these devices. Though, the referenced vulnerabilites require physical acccess from an attacker, so it's up what's more important for you. Also, some people have had success downgrading, resetting BIOS and then updating back to most recent, but we did not.

Steps that worked for us:

0. (optionally) Backup your BIOS preferences especially if you have multiple/custom boot paths set up
1. Enable BIOS downgrade
2. Install the most recent firmware before the update that disabled undervolting via the official installer
    - Note that some parts of the install may fail (for example for us EC firmware downgrade on XPS 9570) but ultimately the BIOS got downgraded successfully
3. As soon as the firmware install finished be ready to boot into BIOS (`F2`) with the first reboot, as otherwise it may be overwritten right away
4. Check that you are indeed on the BIOS version you intended to roll back 
    - If it did not work, the following steps probably won't help either. You may wish to try alternative options below
5. Reset BIOS to factory settings
6. Disable `UEFI Firmware Update Via Capsules in Security` (skipping this step definitely messes up the process on Windows at least)
7. Tell your laptop how much you love it
8. Reboot and hope for the best

Alternative options:

Many people have had success with resetting BIOS via BIOS recovery if they were on the version that disabled undervolting. i.e. if you are on `BIOS version X` which disabled undervolting this could have a high chance of succeeding. But all our devices were on `BIOS X+1` or more recent and this did not help.

0. Backup any BIOS settings
1. Turn off and unplug your laptop
2. Press and hold `ESC` and `Ctrl`
3. Plug your laptop back in and keep holding for about a second, then release
4. In BIOS recovery choose the revert/restore previous version
5. Apply steps __3.__ to __8.__ from the list above

Hope this helps.

moritzschaefer, 2020-11-18:
Thank you @almasen for this small tutorial and even looking for the correct firmware version. A few notes:

 - On Windows, ThrottleStop still seemed to work, despite the new firmware. Hence, although the fixes of the CVEs might affect the capability to Undervolt, there still seems to be a way to undervolt with the new firmware versions
- Regarding step 7: I hate my laptop! This issue is one of several reasons for it.
- Downgrading to 1.18.0 indeed seemed to fix the issue for me.

Again, thank you, and have a nice day

Maverobot, 2020-11-18:
It might be a stupid question. How can I downgrade bios on Linux? I tried with `fwupd` and could not downgrade it:

```
# fwupdmgr downgrade
Choose a device:
0.	Cancel
1.	b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2 (XPS 13 7390 System Firmware)
2.	c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7 (KXG60ZNV512G NVMe KIOXIA 512GB)
1
No downgrades for device, current is 0.1.6.0: 0.1.6.0=same, 0.1.5.1=lowest, 0.1.4.0=lowest, 0.1.3.1=lowest, 0.1.2.0=lowest
```

mensfeld, 2020-11-19:
@almasen @Maverobot in the end I did not have to downgrade bios for my 7390 - it was sufficient to reset things.

Maverobot, 2020-11-19:
> @almasen @Maverobot in the end I did not have to downgrade bios for my 7390 - it was sufficient to reset things.

Thanks. My laptop came out of factory with the newer version bios. I will try resetting it anyway if I cannot find a "cleaner" way to downgrade it.

almasen, 2020-11-19:
>     * Downgrading to 1.18.0 indeed seemed to fix the issue for me.

Really glad it worked!

> 
> @almasen @Maverobot in the end I did not have to downgrade bios for my 7390 - it was sufficient to reset things.

That's great! Strangely, we had no success with resetting, but it's always worth trying first