georgewhewell/undervolt #128: Any plans for Intel 10th gen

luizconrado, 2020-10-07:
Hi,

First of all.
I am a big fan of your work.

Second, I wonder...
Are there any plans to add Intel 10th gen to your project.
I know most laptops have this feature disabled. However, for some brands, like MSI, Clevo laptops, etc. one is allowed to unlock undervolting in the bios.

So, do you plan to add Intel 10th gen undervolting to this project?

Thank you.

elsandosgrande, 2020-10-07:
[My laptop (i5 variant)](https://topics-cdn.dell.com/pdf/inspiron-17-3793-laptop_users-guide_en-us.pdf) has no issues with undervolting:

<details>
<p>

```
sandy@sandys-inspiron:~$ lscpu && sudo undervolt --read
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              2
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           126
Model name:                      Intel(R) Core(TM) i5-1035G1 CPU @ 1.00GHz
Stepping:                        5
CPU MHz:                         1300.019
CPU max MHz:                     3600,0000
CPU min MHz:                     400,0000
BogoMIPS:                        2380.80
Virtualization:                  VT-x
L1d cache:                       192 KiB
L1i cache:                       128 KiB
L2 cache:                        2 MiB
L3 cache:                        6 MiB
NUMA node0 CPU(s):               0-7
Vulnerability Itlb multihit:     KVM: Mitigation: VMX disabled
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Enhanced IBRS, IBPB conditional, RSB filling
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon
                                  pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2a
                                 pic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexprio
                                 rity ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid avx512f avx512dq rdseed adx smap avx512ifma clflushopt intel_pt avx512cd sha_ni avx512bw avx512vl xsaveo
                                 pt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq a
                                 vx512_vnni avx512_bitalg avx512_vpopcntdq rdpid fsrm md_clear flush_l1d arch_capabilities
temperature target: -1 (99C)
core: -65.43 mV
gpu: -75.2 mV
cache: -65.43 mV
uncore: 8000.0 mV
analogio: 0.0 mV
powerlimit: 51.0W (short: 0.00244140625s - enabled) / 22.0W (long: 28.0s - enabled)
```

</p>
</details>

And yes, the values are actually being applied, as the laptop froze a couple of times while dialing the settings in.

georgewhewell, 2020-12-27:
closing as appears to work

closed 2020-12-27 by georgewhewell