georgewhewell/undervolt #135: Followed instructions exactly and this is of course the errror code I get as usual with everything here

TrevorLowe, 2020-12-12:
undervolt --read
modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "/home/owner/.local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/home/owner/.local/lib/python3.8/site-packages/undervolt.py", line 386, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1.

jharwinbarrozo, 2020-12-16:
You have to run with sudo.

`sudo undervolt --read`

trevluv, 2020-12-18:
> You have to run with sudo.
> 
> `sudo undervolt --read`

Now I am getting this error when I followed your advice.

owner@owner-Legion-Y540-15IRH:~$ sudo undervolt --read
sudo: undervolt: command not found

georgewhewell, 2020-12-27:
`sudo python -m undervolt`
or
`sudo $(which undervolt)`

should work

closed 2021-05-01 by georgewhewell