georgewhewell/undervolt #136: Unknown command "undervolt"

gvgndr, 2020-12-13:
`pip list` outputs `undervolt                0.3.0`;
`~$ pip check undervolt
No broken requirements found.`;
but `$ pip help undervolt
ERROR: unknown command "undervolt"`, and
so does `sudo undervolt`.
I'm using ubuntu 20.10.

MariusSp, 2020-12-23:
Did you install undervolt for the user or root?

I assume you installed it in your userspace. Please check if your user python /bin is part of your $PATH
`echo $PATH`
should contain 
`/home/<user>/.local/bin`

gvgndr, 2020-12-27:
> Did you install undervolt for the user or root?
> 
> I assume you installed it in your userspace. Please check if your user python /bin is part of your $PATH
> echo $PATH
> should contain
> /home/<user>/.local/bin
> or whereever your python environment installs its packages to.

Yes, indeed, that is the output I get. In fact, `~$ undervolt` works as expected,
but with `$ undervolt --read` I get
```
modprobe: ERROR: could not insert 'msr': Operation not permitted
Traceback (most recent call last):
  File "/home/andrea/.local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/home/andrea/.local/bin/undervolt.py", line 386, in main
    subprocess.check_call(['modprobe', 'msr'])
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['modprobe', 'msr']' returned non-zero exit status 1.
```

What does this mean? Also, would it be better to install undervolt as root?
Thank you!

MariusSp, 2020-12-27:
I'm certain it would be easier if you installed it globally.

closed 2021-05-01 by georgewhewell