georgewhewell/undervolt #137: Undervolt not working on i5-1135G7 (Dell XPS 9310 DE)

abraha2d, 2021-01-08:
Getting this error when running `sudo undervolt --core -100 --cache -100 -v`:
```
$ sudo undervolt --core -100 --cache -100 -v
INFO:root:Setting core offset to -100mV
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/0/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/1/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/2/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/3/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/4/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/5/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/6/msr
INFO:root:Writing 0x80000011f3400000 to /dev/cpu/7/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/0/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/1/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/2/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/3/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/4/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/5/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/6/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/7/msr
INFO:root:Read 0x1f00000000 from /dev/cpu/0/msr
ERROR:root:Failed to apply core: set -99.609375, read 60000.0
```
This is what `sudo undervolt -r` says:
```
$ sudo undervolt -r
temperature target: -0 (100C)
core: 60000.0 mV
gpu: 60000.0 mV
cache: 60000.0 mV
uncore: 60000.0 mV
analogio: 60000.0 mV
powerlimit: 64.0W (short: 0.00244140625s - disabled) / 64.0W (long: 28.0s - enabled)
```
These two lines are logged to dmesg when running undervolt (doesn't matter whether setting or just reading):
```
[  889.462018] filter_write: 15 callbacks suppressed
[  889.462020] msr: Write to unrecognized MSR 0x150 by undervolt (pid: 6828). Please report to x86@kernel.org.
```
uname: `Linux 5.10.0-1008-oem #9-Ubuntu SMP Tue Dec 15 14:22:38 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
Installed undervolt with `sudo python3 -m pip uninstall undervolt`.

georgewhewell, 2021-01-10:
you can try adding `msr.allow_writes=on` to your kernel commandline. userspace applications (even as root) are being blocked from accessing these msr in new kernel versions :/

abraha2d, 2021-01-23:
Tried adding `msr.allow_writes=on` to the kernel command line, but I still get the same results. The only difference is that it no longer prints those logs to `dmesg`.

Areidz, 2021-04-04:
I've read somewhere that it's locked in these processors because of plundervolt vulnerability. I couldn't get it working on my MSI Modern 14.

24fpsDaVinci, 2022-01-17:
same problem on xps 9310 with fedora 35, kernel 5.16

papayayooo, 2022-08-03:
FIXME but I think 11th/12th gen Intel CPUs don't have undervolt enabled anymore due to Plundervolt (or other reasons). I wanted to buy a 12th-gen, but they are very power hungry so I wanted to use undervolt heavily. Turns out, you can't do that anymore.

wrobelda, 2023-12-10:
FOr the record, motherboards manufacturers can choose to enable undervolting for 12th gen processors onwards. To check if one's motherboard is affected, one can check whether the "Undervolt Protection" with:
```
rdmsr -f 1:1 0x195
```
1 menas protection is enabled, 0 means it's disabled.

aboode95, 2024-01-17:
> one can check whether the "Undervolt Protection" is enabled

@wrobelda, so if "Undervolt Protection" is disabled, I can undervolt?