georgewhewell/undervolt #138: Undervolt not working on Dell XPS 7590

BRA1L0R, 2021-01-13:
Yet another dell xps issue,

the XPS 15 7590 has been added to the support list in the latest commit but undervolt doesn't seem to be working on my machine. I tried both 1.5.1 and the latest bios (1.8.1).

jayv, 2021-01-14:
same issue here XPS 7590
```
sudo python undervolt.py --core -150                                                                
WARNING:root:You have supplied different offsets for Core and Cache. The smaller of the two (or none if you only supplied one) will be applied to both planes.
ERROR:root:Failed to apply core: set -150.390625, read 0.0
```

hackthejawn, 2021-02-12:
Having the same issue on Linux Mint 20.1. Command and results:

`sudo undervolt --core -100 --cache -100
ERROR:root:Failed to apply core: set -99.609375, read 0.0`

Willing to help with diagnostics and testing if you need anything.

hackthejawn, 2021-02-12:
Fixed this by resetting BIOS to factory defaults, per this issue:

https://github.com/kitsunyan/intel-undervolt/issues/43#issuecomment-618412498