georgewhewell/undervolt #151: Intel i5-3470 I/O Errors

d1nuc0m, 2021-04-27:
Installed undervolt (Intel i5-3470, Kubuntu 20.04) with _sudo pip3 install undervolt_.
When trying to use it, I always get Input/Output errors:

```
sudo undervolt --read
temperature target: -0 (100C)
Traceback (most recent call last):
  File "/usr/local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 445, in main
    voltage = read_offset(plane, msr)
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 204, in read_offset
    write_msr(value_to_write, msr.addr_voltage_offsets)
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 79, in write_msr
    os.write(f, pack('Q', val))
OSError: [Errno 5] Input/output error
```
```
sudo undervolt --core -5 --cache -5 
Traceback (most recent call last):
  File "/usr/local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 401, in main
    set_offset(plane, offset, msr)
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 218, in set_offset
    write_msr(write_value, msr.addr_voltage_offsets)
  File "/usr/local/lib/python3.8/dist-packages/undervolt.py", line 79, in write_msr
    os.write(f, pack('Q', val))
OSError: [Errno 5] Input/output error
```

georgewhewell, 2021-05-01:
probably secure boot enabled, try to disable

d1nuc0m, 2021-05-01:
> probably secure boot enabled, try to disable

No, it's already been disabled (checked twice)