georgewhewell/undervolt #154: failed run at systemd

biladina, 2021-05-02:
I try to run undervolt in console, its run perfectly.. but when I try to run it via systemd, it fails with error Permission denied..
tried to googling it and people says just add user/group with `root`, but it won't help..

this is the log when try to start service :
```
× undervolt.service - undervolt
     Loaded: loaded (/etc/systemd/system/undervolt.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2021-05-02 19:55:38 WIB; 1min 50s ago
    Process: 17055 ExecStart=/usr/local/bin/undervolt -v --core -150 --cache -150 --gpu -100 (code=exited, status=1/FAILURE)
   Main PID: 17055 (code=exited, status=1/FAILURE)
        CPU: 54ms

May 02 19:55:38 localhost.localdomain undervolt[17055]:   File "/usr/local/lib/python3.9/site-packages/undervolt.py", line 401, in main
May 02 19:55:38 localhost.localdomain undervolt[17055]:     set_offset(plane, offset, msr)
May 02 19:55:38 localhost.localdomain undervolt[17055]:   File "/usr/local/lib/python3.9/site-packages/undervolt.py", line 218, in set_offset
May 02 19:55:38 localhost.localdomain undervolt[17055]:     write_msr(write_value, msr.addr_voltage_offsets)
May 02 19:55:38 localhost.localdomain undervolt[17055]:   File "/usr/local/lib/python3.9/site-packages/undervolt.py", line 79, in write_msr
May 02 19:55:38 localhost.localdomain undervolt[17055]:     os.write(f, pack('Q', val))
May 02 19:55:38 localhost.localdomain undervolt[17055]: PermissionError: [Errno 13] Permission denied
May 02 19:55:38 localhost.localdomain systemd[1]: undervolt.service: Main process exited, code=exited, status=1/FAILURE
May 02 19:55:38 localhost.localdomain systemd[1]: undervolt.service: Failed with result 'exit-code'.
May 02 19:55:38 localhost.localdomain systemd[1]: Failed to start undervolt.
```

and this is my service file :
```
[Unit]
Description=undervolt
After=suspend.target
After=hibernate.target
After=hybrid-sleep.target

[Service]
User=root
Group=root
Type=oneshot
# If you have installed undervolt globally (via sudo pip install):
ExecStart=/usr/local/bin/undervolt -v --core -150 --cache -150 --gpu -100
# If you want to run from source:
# ExecStart=/path/to/undervolt.py -v --core -150 --cache -150 --gpu -100

[Install]
WantedBy=multi-user.target
WantedBy=suspend.target
WantedBy=hibernate.target
WantedBy=hybrid-sleep.target
```

machine : thinkpad p52
OS : Fedora 34 with KDE

matthtan, 2021-05-02:
Same problem here, switched from Pop OS (Ubuntu) and now having this problem.
machine: thinkpad p53
OS: Fedora 34 Gnome

biladina, 2021-05-10:
its strange, somehow after getting update from fedora, it works, running smoothly..

maybe there is some workaround about SELinux in Fedora 34 which is still new.. and after some update, it fixed..

closed 2021-05-10 by biladina