georgewhewell/undervolt #157: Write to unrecognised MSR 0x150

ldaws011, 2021-06-30:
dmesg prints: 
```
[ 2403.691425] filter_write: 71 callbacks suppressed
[ 2403.691426] msr: Write to unrecognized MSR 0x150 by python3 (pid: 17342). Please report to x86@kernel.org.
```
upon running undervolt.py for adjusting (--core, --cache) and reading (--read) voltages
Running on an Acer Nitro 5 with i5-8300H CPU.
I believe the CPU still undervolts, and --read prints correct information, so I'm assuming this warning can be safely ignored?

GFNIAE, 2021-11-11:
Hello,
I meet the same error message on kernel log :
"Write to unrecognized MSR 0x150 by undervolt (pid: 2423). Please report to x86@kernel.org."
According to undervolt --read, the settings are applied, but this message seems like there is at least one action of the script that does not reach its goal.
Is there something to check or to try ?

georgewhewell, 2021-11-11:
it's just a warning, can ignore it

closed 2021-11-11 by georgewhewell

GFNIAE, 2021-11-12:
Hello geordewhewell, thanks for your anwser !
I picked in another of your answer this trial : I add "msr.allow_writes=on" on the kernel parameter. As a consequence there is no warning anymore in dmesg. Do you know if it's better not to use this parameter (and deal with the dmesg warning), or if it does not hurt ?