georgewhewell/undervolt #161: PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

kachapman, 2021-09-14:
When trying to use the program, I get this output:

> USER:~$ undervolt -r
Traceback (most recent call last):
  File "/home/kenchapman/.local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/home/kenchapman/.local/lib/python3.9/site-packages/undervolt.py", line 439, in main
    temp = read_temperature(msr)
  File "/home/kenchapman/.local/lib/python3.9/site-packages/undervolt.py", line 191, in read_temperature
    return (read_msr(msr.addr_temp) & (127 << 24)) >> 24
  File "/home/kenchapman/.local/lib/python3.9/site-packages/undervolt.py", line 88, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

georgewhewell, 2021-09-18:
use `sudo`

kachapman, 2021-09-18:
I forgot to mention that Sudo doesn't fix the issue.

-------- Original Message --------

kachapman, 2021-10-04:
> ken-sb2:~$ sudo undervolt -r
> sudo: undervolt: command not found
>

kachapman, 2021-10-04:
Looks like the issue was secure boot all along. Is there no way to enable this with secure boot enabled?

closed 2021-10-04 by kachapman