georgewhewell/undervolt #163: Does not work with thinkpad x1/p1 gen4

BonaBeavis, 2021-10-16:
<pre><font color="#FF0086">❯</font> <font color="#1FAAAA"><b>~</b></font> <font color="#00C918">sudo</font> <font color="#00C918">undervolt</font> <font color="#1FAAAA">-v</font> <font color="#1FAAAA">--core</font> <font color="#1FAAAA">-15</font> <font color="#1FAAAA">--cache</font> <font color="#1FAAAA">-15</font> 
INFO:root:Setting core offset to -15mV
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/0/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/1/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/2/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/3/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/4/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/5/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/6/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/7/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/8/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/9/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/10/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/11/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/12/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/13/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/14/msr
INFO:root:Writing 0x80000011fe200000 to /dev/cpu/15/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/0/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/1/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/2/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/3/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/4/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/5/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/6/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/7/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/8/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/9/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/10/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/11/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/12/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/13/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/14/msr
INFO:root:Writing 0x8000001000000000 to /dev/cpu/15/msr
INFO:root:Read 0x0 from /dev/cpu/0/msr
ERROR:root:Failed to apply core: set -14.6484375, read 0.0</pre>

kylebakerio, 2021-11-16:
literally came here just now for this. damn.

I think I read they aren't able to be undervolted, because there's no known mitigation for plundervolt? can this processor be undervolted on windows? (I'm on the i9, fwiw, also x1 extreme gen 4)

BonaBeavis, 2021-11-16:
Looks like it, there is a glimmer of hope, because plundervolt is no issue when intel sgx is deactivated. Lenovo could enable undervolting per firmware update.

SuchHate, 2022-04-13:
I guess there is nothing new? Revceived my P1 gen 4, undervolt would do wonders on thermals and battery life.

kylebakerio, 2022-04-13:
>  plundervolt is no issue when intel sgx is deactivated. 

would love to see a source for this. I _definitely_ want undervolting more than intel sgx.