georgewhewell/undervolt #166: Undervolt service failed

xQzca, 2022-01-27:
**Running 'systemctl start undervolt' shows this:**

`Job for undervolt.service failed because the control process exited with error code.
See "systemctl status undervolt.service" and "journalctl -xeu undervolt.service" for details.`

**Running 'systemctl status undervolt.service shows this:**

`× undervolt.service - undervolt
     Loaded: loaded (/etc/systemd/system/undervolt.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Fri 2022-01-28 10:22:48 AEDT; 2min 46s ago
    Process: 4287 ExecStart=/usr/local/bin/undervolt -v --core -100 --cache -100 --gpu -50 --uncore -50 --temp 92 (code=exited, status=203/EXEC)
   Main PID: 4287 (code=exited, status=203/EXEC)
        CPU: 635us

Jan 28 10:22:48 Unknown systemd[1]: Starting undervolt...
Jan 28 10:22:48 Unknown systemd[4287]: undervolt.service: Failed to locate executable /usr/local/bin/undervolt: No such file or directory
Jan 28 10:22:48 Unknown systemd[4287]: undervolt.service: Failed at step EXEC spawning /usr/local/bin/undervolt: No such file or directory
Jan 28 10:22:48 Unknown systemd[1]: undervolt.service: Main process exited, code=exited, status=203/EXEC
Jan 28 10:22:48 Unknown systemd[1]: undervolt.service: Failed with result 'exit-code'.
Jan 28 10:22:48 Unknown systemd[1]: Failed to start undervolt.`

**And running 'journalctl -xeu undervolt.service' shows this:**

`Jan 28 10:22:48 Unknown systemd[1]: undervolt.service: Main process exited, code=exited, status=203/EXEC
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStart= process belonging to unit undervolt.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 203.
Jan 28 10:22:48 Unknown systemd[1]: undervolt.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit undervolt.service has entered the 'failed' state with result 'exit-code'.
Jan 28 10:22:48 Unknown systemd[1]: Failed to start undervolt.
░░ Subject: A start job for unit undervolt.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit undervolt.service has finished with a failure.
░░ 
░░ The job identifier is 1562 and the job result is failed.`

OS: EndeavourOS
DE: Gnome
CPU: i5-9300H
GPU: GTX 1650 Mobile

Any way to fix this? This worked on my previous install of Pop!_OS 21.10.

georgewhewell, 2022-01-28:
the important line is- 
```
/usr/local/bin/undervolt: No such file or directory
```

you need to find where the file is installed and change the path to that- `which undervolt` should tell you

xQzca, 2022-01-29:
thanks, this fixed it. it was in /usr/bin/undervolt

closed 2022-01-29 by xQzca