georgewhewell/undervolt #167: PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

gmnpjpn, 2022-02-03:
I installed undervolt and the command "undervolt" worked fine, when I tried "undervolt --r" I received that output on the terminal:

Traceback (most recent call last):
  File "/home/german/.local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/home/german/.local/lib/python3.8/site-packages/undervolt.py", line 439, in main
    temp = read_temperature(msr)
  File "/home/german/.local/lib/python3.8/site-packages/undervolt.py", line 191, in read_temperature
    return (read_msr(msr.addr_temp) & (127 << 24)) >> 24
  File "/home/german/.local/lib/python3.8/site-packages/undervolt.py", line 88, in read_msr
    f = os.open(n, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/cpu/0/msr'

I have secure boot disabled.

georgewhewell, 2022-02-03:
you need to run as su-  `sudo undervolt -r`

gmnpjpn, 2022-02-03:
I tried but I receive that output:

sudo: undervolt: command not found

georgewhewell, 2022-02-03:
you can try `sudo -E undervolt` or `sudo $(which undervolt)`

gmnpjpn, 2022-02-03:
With the first command I get that output:

sudo: undervolt: command not found

With the second:

usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
            [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-T timeout] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-T timeout] [-u user] file ...

georgewhewell, 2022-02-03:
the output of the second command implies undervolt is not on your `PATH`.. weird.. anyway, it says in your first comment the script is at `/home/german/.local/lib/python3.8/site-packages/undervolt.py`, so just `sudo python /home/german/.local/lib/python3.8/site-packages/undervolt.py` should work

gmnpjpn, 2022-02-03:
I tried that, the output and verified the directory is in the path:

When I run `echo $PATH`
~/.local/lib/python3.8/site-packages:/home/german/.local/bin:/home/german/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

When I run `sudo python /home/german/.local/lib/python3.8/site-packages/undervolt.py`:

sudo: python: command not found

sudharsan2000, 2022-04-22:
> When I run `sudo python /home/german/.local/lib/python3.8/site-packages/undervolt.py`:
> 
> sudo: python: command not found

Can you try using python3?
`sudo python3 /home/german/.local/lib/python3.8/site-packages/undervolt.py`

IceCereal, 2022-09-29:
A (generally inadvisable) alternative would be to install undervolt with sudo, i.e. `sudo pip3 install undervolt`.

CygnusTerminal, 2023-03-27:
Same issue here
`fsamur@asus-gl552vw:~$ sudo $(which undervolt)`

`Traceback (most recent call last):
  File "/home/fsamur/.local/bin/undervolt", line 5, in <module>
    from undervolt import main
ModuleNotFoundError: No module named 'undervolt'
`