georgewhewell/undervolt #169: Not working on mint 20

Cloud-Mak, 2022-03-12:
So I am getting command not found.

```
makrand@zmint:~$ pip show undervolt
Name: undervolt
Version: 0.3.0
Summary: Undervolt Intel CPUs under Linux
Home-page: http://github.com/georgewhewell/undervolt
Author: George Whewell
Author-email: georgerw@gmail.com
License: GPL
Location: /home/makrand/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
makrand@zmint:~$ pip uninstall undervolt
Found existing installation: undervolt 0.3.0
Uninstalling undervolt-0.3.0:
  Would remove:
    /home/makrand/.local/bin/undervolt
    /home/makrand/.local/lib/python3.8/site-packages/undervolt-0.3.0.dist-info/*
    /home/makrand/.local/lib/python3.8/site-packages/undervolt.py
Proceed (y/n)? n

```
Treid doing w and w/0 sudo. no luck. What might be wrong?

```
makrand@zmint:~$ undervolt
undervolt: command not found
makrand@zmint:~$ sudo undervolt
[sudo] password for makrand:            
sudo: undervolt: command not found
```

seifhassine, 2022-06-15:
Same issue on Ubuntu 20.04.4 LTS

Cloud-Mak, 2022-06-16:
> Same issue on Ubuntu 20.04.4 LTS

Looks like this project is dead!

Ahmad791, 2022-06-27:
Fixed the problem 
it appears to be that there is a problem with locating the package
you need to search for "undervolt.py", mine was in "/anaconda3/lib/python3.8/site-packages" then right click in folder and choose "open in terminal" (can navigate there using terminal too), then run "sudo python undervolt.py" in terminal followed by the command you want. 
For example "sudo python undervolt.py -r"

queenbiscuit311, 2023-05-11:
for me it was just that ~/.local/bin wasnt in PATH and after adding it the command worked