georgewhewell/undervolt #177: You need to have root privileges to run this script with these options.

jacekgajek, 2023-03-29:
I've run the following commands:
Installation:
```
(base) jacek@cammembert:~$ sudo apt install python3-pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libkf5waylandserver5 libkwaylandserver5 linux-hwe-5.19-headers-5.19.0-32
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libexpat1-dev libpython3-dev libpython3.10-dev python3-dev python3-wheel python3.10-dev
The following NEW packages will be installed:
  libexpat1-dev libpython3-dev libpython3.10-dev python3-dev python3-pip python3-wheel python3.10-dev
0 upgraded, 7 newly installed, 0 to remove and 7 not upgraded.
Need to get 6 780 kB of archives.
After this operation, 29,7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://pl.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libexpat1-dev amd64 2.4.7-1ubuntu0.2 [147 kB]
Get:2 http://pl.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpython3.10-dev amd64 3.10.6-1~22.04.2ubuntu1 [4 755 kB]
Get:3 http://pl.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpython3-dev amd64 3.10.6-1~22.04 [7 166 B]             
Get:4 http://pl.archive.ubuntu.com/ubuntu jammy-updates/main amd64 python3.10-dev amd64 3.10.6-1~22.04.2ubuntu1 [507 kB]     
Get:5 http://pl.archive.ubuntu.com/ubuntu jammy-updates/main amd64 python3-dev amd64 3.10.6-1~22.04 [26,0 kB]                
Get:6 http://pl.archive.ubuntu.com/ubuntu jammy-updates/universe amd64 python3-wheel all 0.37.1-2ubuntu0.22.04.1 [32,0 kB]   
Get:7 http://pl.archive.ubuntu.com/ubuntu jammy-updates/universe amd64 python3-pip all 22.0.2+dfsg-1ubuntu0.2 [1 305 kB]     
Fetched 6 780 kB in 7s (913 kB/s)                                                                                            
Selecting previously unselected package libexpat1-dev:amd64.
(Reading database ... 531494 files and directories currently installed.)
Preparing to unpack .../0-libexpat1-dev_2.4.7-1ubuntu0.2_amd64.deb ...
Unpacking libexpat1-dev:amd64 (2.4.7-1ubuntu0.2) ...
Selecting previously unselected package libpython3.10-dev:amd64.
Preparing to unpack .../1-libpython3.10-dev_3.10.6-1~22.04.2ubuntu1_amd64.deb ...
Unpacking libpython3.10-dev:amd64 (3.10.6-1~22.04.2ubuntu1) ...
Selecting previously unselected package libpython3-dev:amd64.
Preparing to unpack .../2-libpython3-dev_3.10.6-1~22.04_amd64.deb ...
Unpacking libpython3-dev:amd64 (3.10.6-1~22.04) ...
Selecting previously unselected package python3.10-dev.
Preparing to unpack .../3-python3.10-dev_3.10.6-1~22.04.2ubuntu1_amd64.deb ...
Unpacking python3.10-dev (3.10.6-1~22.04.2ubuntu1) ...
Selecting previously unselected package python3-dev.
Preparing to unpack .../4-python3-dev_3.10.6-1~22.04_amd64.deb ...
Unpacking python3-dev (3.10.6-1~22.04) ...
Selecting previously unselected package python3-wheel.
Preparing to unpack .../5-python3-wheel_0.37.1-2ubuntu0.22.04.1_all.deb ...
Unpacking python3-wheel (0.37.1-2ubuntu0.22.04.1) ...
Selecting previously unselected package python3-pip.
Preparing to unpack .../6-python3-pip_22.0.2+dfsg-1ubuntu0.2_all.deb ...
Unpacking python3-pip (22.0.2+dfsg-1ubuntu0.2) ...
Setting up python3-wheel (0.37.1-2ubuntu0.22.04.1) ...
Setting up libexpat1-dev:amd64 (2.4.7-1ubuntu0.2) ...
Setting up libpython3.10-dev:amd64 (3.10.6-1~22.04.2ubuntu1) ...
Setting up python3-pip (22.0.2+dfsg-1ubuntu0.2) ...
Setting up python3.10-dev (3.10.6-1~22.04.2ubuntu1) ...
Setting up libpython3-dev:amd64 (3.10.6-1~22.04) ...
Setting up python3-dev (3.10.6-1~22.04) ...
Processing triggers for man-db (2.10.2-1) ...
```

Then

```
pip install undervolt
```
Then
```
(base) jacek@cammembert:~$ undervolt -r
You need to have root privileges to run this script with these options.
Rerun with 'sudo'.
(base) jacek@cammembert:~$ sudo undervolt -r
sudo: undervolt: command not found
```
So now I'm in this infinite a loop.

doxxedd, 2023-04-06:
i have the same issue

D-VR, 2023-04-27:
The reason it doesn't work is because under volt is only installed for the user.

A dirty fix would be to run the pip install using sudo.

notstarboard, 2024-01-14:
I ran into this same issue and spent way too long looking for a solution, but it's actually easy. Just install with: 

`pip3 install undervolt`

Then you can run the python package directly from the folder it installed to using sudo. For me it's:

`sudo python3 ~/.local/lib/python3.10/site-packages/undervolt.py`

momincks, 2024-03-02:
install with sudo su
```
sudo su
pip install undervolt
undervolt --read
```