georgewhewell/undervolt #179: ERROR:root:Failed to apply - undervolt settings on Intel Celeron N5100 and Clear Linux

LaurenceGough, 2023-05-26:
Hello all,

I thought I'd have a crack at this. I can read my current settings on my Intel Celeron N5100 but I am unable to set any values at all on gpu, cores etc.

The error is for example:
```
sudo undervolt --gpu -30
ERROR:root:Failed to apply gpu: set -30.2734375, read 0.0
```

I am using Clear Linux. Would I need to follow these steps? "Note for custom kernel users: the module 'msr' ("CONFIG_X86_MSR") must be enabled in the kernel options (Processor type & features -> Model-specific register support)"

I haven't made or edited the kernel at all, is there any way to check if Clear Linux supports MSR?

Or perhaps my BIOS or CPU microcode has the plundervolt patch applied? It's a Chinese mini PC BIOS.

Many thanks

georgewhewell, 2023-06-21:
i don't think adjusting FIVR is supported on celeron cpu, i don't see anyone confirming in the support list: https://github.com/georgewhewell/undervolt#hardware-support

closed 2023-06-21 by georgewhewell

LaurenceGough, 2023-06-27:
Update: I enabled MSR editing via Kernel boot commands, and turned off CPU lock in BIOS and also tried a BIOS factory reset but sadly the same result happens. I am on the latest BIOS update available. 

In regards to FIVR, I found this from Intel:

**Fully Integrated Voltage Regulator (FIVR)**
The processor integrates multiple voltage rails in order to reduce BOM costs for the
platform, and to enable additional voltage level features the processor can take
advantage of.
There are FIVRs integrated on the PCH, VNN/V1P05 which is sourced from
VCCIN_Aux. VCCIN_Aux also sourced the VCCSA rail in a CPU. In addition to VCCSA
FIVR, compute die integrates 4 additional FIVRs to source VCCCORE, VCCSA, VCCL2,
VCCGT and VCCRING, which derives the respective voltages from VCCIN VR on
platform. Each FIVR is able to control a specific voltage rail.

**Source:**
Intel® Pentium® Silver and Intel®Celeron®
Processors
Datasheet
https://cdrdv2.intel.com/v1/dl/getcontent/633935 - https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/servers/platforms/intel-pentium-silver-and-intel-celeron-processors-datasheet-volume-1-of-2/006/skus/

I also emailed Intel but they are just doing the standard copy and paste reply. 

I have enabled full debug options in my BIOS but I cannot see anything else to do with locked voltages or the FIVR feature.

I do wonder if it works in Windows, I didn't have much luck finding a live "CD", especially one that wasn't massively cut down. I'll have another look soon.

If any other N5100 / Intel Celeron owners want to give it a try on Windows or Linux any feedback would be much appreciated!